ethz-asl/kalibr #324: Problem with importing files from different folder

KellyYutongHe, 2019-11-08:
When I execute kalibr_calibrate_cameras --h, I got the following error message:

importing libraries
Traceback (most recent call last):
  File "/home/whil/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/whil/kalibr_workspace/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 6, in <module>
    import sm
  File "/home/whil/kalibr_workspace/devel/lib/python2.7/dist-packages/sm/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 4, in <module>
ImportError: /home/whil/kalibr_workspace/devel/lib/libsm_logging.so: undefined symbol: _ZNK5boost16re_detail_10650131cpp_regex_traits_implementationIcE9transformEPKcS4_

Does anyone know how to fix it?

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

goldbattle, 2022-05-19:
Hi, please try the latest version and reopen if you still have an issue, thanks!

closed 2022-05-19 by goldbattle