ethz-asl/kalibr #325: kalibr_calibrate_cameras  ... --show-extraction failed

jyzhong, 2019-11-11:
with --show -extraction . I got "Attempt to unlock mutex that was not locked". Without it, kalibr_calibrate_cameras works.

frme92, 2019-11-25:
I have the same problem, someone can fix this?

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

jyzhong, 2020-01-12:
> execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341
> 
> Thanks very much!

use the tool "kalibr_allan"

jyzhong, 2020-01-18:
try to replace "cvStartWindowThread()" with "cv::waitKey(1)",  it will fix this issue.

closed 2020-01-18 by jyzhong

Xiangyf18, 2021-01-20:
> try to replace "cvStartWindowThread()" with "cv::waitKey(1)", it will fix this issue.
Thanks a lot  for your suggestion ,it really works. 
 Exactly I change  this file : 
 {my_Kalibr_source_code_path}/aslam_cv/aslam_cameras/src/GridCalibrationTargetCheckerboard.cpp

AntarcticaSkj, 2021-12-12:
> 

Thank U! It help me a lot !

AntarcticaSkj, 2021-12-12:
> Thanks a lot for your suggestion ,it really works. Exactly I change this file : {my_Kalibr_source_code_path}/aslam_cv/aslam_cameras/src/GridCalibrationTargetCheckerboard.cpp
> 
> > try to replace "cvStartWindowThread()" with "cv::waitKey(1)", it will fix this issue.

Thank U! It help me a lot !