ethz-asl/kalibr #326: How can I use a checkerboadr to get a good calibration result?

EmiyaEstelle, 2019-11-13:
Hi, guys
I am going to calibrate my camera model with a mems IMU by Kalibr. I have two type of calibration board, one of which is a checkerboard in 0.01 mm accuracy; the other is a April grid and it is a A4 paper printed.
I find the checkerboard is not good for calibration.  In the calibraiton between IMU and camera, it is always to say that

> Optimization problem initialized with 10496 design variables and 74489 error terms
The Jacobian matrix is 169946 x 47212
[0.0]: J: 401996
Exception in thread block: [aslam::Exception] /home/cyx/cali_catkin/src/kalibr-master/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [1.57363e+09 <= 1.57363e+09 < 1.57363e+09]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
[ERROR] [1573634275.844425]: Optimization failed!
Traceback (most recent call last):
  File "/home/cyx/cali_catkin/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/cyx/cali_catkin/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/cyx/cali_catkin/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 206, in main
    iCal.optimize(maxIterations=parsed.max_iter, recoverCov=parsed.recover_cov)
  File "/home/cyx/cali_catkin/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 179, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!
I am sure that I move the camera very and very slowly. But in the same condition, the April grid is good. 
I just have one board in a high accuracy and good quality. So who can teach me how to use the checkerboard to do the good calibration? Thank you!

gaozhiye, 2019-12-06:
> Hi, guys
> I am going to calibrate my camera model with a mems IMU by Kalibr. I have two type of calibration board, one of which is a checkerboard in 0.01 mm accuracy; the other is a April grid and it is a A4 paper printed.
> I find the checkerboard is not good for calibration. In the calibraiton between IMU and camera, it is always to say that
> 
> > Optimization problem initialized with 10496 design variables and 74489 error terms
> > The Jacobian matrix is 169946 x 47212
> > [0.0]: J: 401996
> > Exception in thread block: [aslam::Exception] /home/cyx/cali_catkin/src/kalibr-master/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [1.57363e+09 <= 1.57363e+09 < 1.57363e+09]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
> > [ERROR] [1573634275.844425]: Optimization failed!
> > Traceback (most recent call last):
> > File "/home/cyx/cali_catkin/devel/bin/kalibr_calibrate_imu_camera", line 15, in 
> > exec(compile(fh.read(), python_script, 'exec'), context)
> > File "/home/cyx/cali_catkin/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in 
> > main()
> > File "/home/cyx/cali_catkin/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 206, in main
> > iCal.optimize(maxIterations=parsed.max_iter, recoverCov=parsed.recover_cov)
> > File "/home/cyx/cali_catkin/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 179, in optimize
> > raise RuntimeError("Optimization failed!")
> > RuntimeError: Optimization failed!
> > I am sure that I move the camera very and very slowly. But in the same condition, the April grid is good.
> > I just have one board in a high accuracy and good quality. So who can teach me how to use the checkerboard to do the good calibration? Thank you!

Hi guys!
Can you give me an Aprilgrid calibration target in PDF style?
Thank you very much!

EmiyaEstelle, 2019-12-06:
@gaozhiye 
Hello. You can create the PDF with the commands 
`kalibr_create_target_pdf --type apriltag --nx [NUM_COLS] --ny [NUM_ROWS] --tsize [TAG_WIDTH_M] --tspace [TAG_SPACING_PERCENT]`. I am the sourece codes install. I am not sure whether the CDE can do this.
You can see the course in kalibr wiki.  Maybe you will meet the same question to fail to create the PDF and you can see https://github.com/ethz-asl/kalibr/issues/92

MrHusky007, 2019-12-07:
> @gaozhiye
> Hello. You can create the PDF with the commands
> `kalibr_create_target_pdf --type apriltag --nx [NUM_COLS] --ny [NUM_ROWS] --tsize [TAG_WIDTH_M] --tspace [TAG_SPACING_PERCENT]`. I am the [...]

Hello!
When I typed  **kalibr_create_target_pdf --type apriltag --nx 6 --ny 6 --tsize 0.088 --tspace 0.3**
in the command line ,the reply is ,can not find the command .
So ,can you give me a calibration target in PDF format at your convenience ?
Thank you very much!

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

MrHusky007, 2020-01-12:
You can search imu calibrate tool , the imu must calibrated seprarately.

EmiyaEstelle, 2020-01-13:
@MrHusky007 
Thanks for your help. I see #341 and I think that it seems that I have not expressed my problem. I have a model and it consists of an IMU and a binocular camera. Before I use Kalibr to get the transformation matrix between IMU and camera, I have got the IMU error model (I get the IMU YAML)by another tool.
Now I meet the problem is that it will be given a worse result even though my check board is in a high accuracy and the optimizer can not work because of non-convergence, while the April grid target is good for imu-cam calibration although the target is printed in an A-4 paper. I would like to know how to get a good result by using the check board target. Thanks!

MrHusky007, 2020-01-13:
https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration

NCHUPJ, 2020-05-19:
@EmiyaEstelle I have encountered the same problem. How did you solve it? Thank you very much.

EmiyaEstelle, 2020-05-25:
@NCHUPJ 
Hi. I can only give you some suggestions about my failure experiences. I can not make sure whether it will help you to get a good result.
First, whatever the type of checkboard you use, it must be a flat board. In the earliest time, I printed the checkboard to an A4 paper and paste it to the wall. It could not get a good result. Then I bought a specific board and the result improved. By the way, I saw someone said that he used the screen to show the checkboard. Maybe you can try.
Others, in my experiences, the chess checkboard is not good at the camera-imu calibration only in Kailibr. I guess that the inuput data is recorded in a dynamic procession. Maybe you can get a not bad result in camera calibration. The April Grid (or called AprilTag) is good there.

oulton, 2021-04-08:
@EmiyaEstelle  I have encountered the same problem. 
File "/home/hyper/code/catkin_tools/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 179, in optimize
    raise RuntimeError("Optimization failed!")
RuntimeError: Optimization failed!

I try to record many bags to let sensors move slowly,but it doesn't work.
Can you tell me how did you solve it? And what  causes it?

EmiyaEstelle, 2021-05-10:
@oulton 
Hi. I have not done the calibration for some time and I am not sure that my suggestion can help you there.
First and the important thing is that I think the AprilTag board is better than Chess board. I remembered that I failed the Chess board in the calibration of relationship between IMU and camera. The same thing like you is that the optimization failed.
The second is that I buy a glassic and Al2O3 board to do the calibration. It is much better than the "board" printed in to a papper and pasted on the wall.

goldbattle, 2022-05-19:
We recommend the apriltag grid since it resolves the ambiguity problem and allows for better partial detections.

closed 2022-05-19 by goldbattle