ethz-asl/kalibr #327: Multi IMU and IMU intrinsic calibration

AlyonaV, 2019-11-21:
According to Wiki, kalibr "allows for estimating IMU intrinsics as well as the displacement of the accelerometer y- and z-axis with respect to its x-axis". I run kalibr_calibrate_imu_camera but can`t find out how to get the value of the displacement of the accelerometer y- and z-axis with respect to its x-axis. I suppose that this should be written in the file imu-%BAGNAME%.yaml. But I have just this data:
```
imu0:
  T_i_b:
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
  accelerometer_noise_density: 0.02
  accelerometer_random_walk: 0.002
  gyroscope_noise_density: 0.02
  gyroscope_random_walk: 4.08068016281e-05
  model: calibrated
  rostopic: /imu0
  time_offset: 0.0
  update_rate: 400.0
```

How can I get the the value of the displacement of the accelerometer y- and z-axis with respect to its x-axis?

goldbattle, 2022-07-08:
You need an aiding sensor. In the case of Kalibr you will need a camera along with any IMUs you wish to calibrate.
Thanks for answering OPs question, but in the future open a new issue if you have a question that isn't the same as the original poster, thanks!

closed 2022-07-08 by goldbattle

alvaro-oliver, 2022-07-08:
I'm sorry for the offtopics.
Thank you for the quick answer and thanks for updating the wiki also.