ethz-asl/kalibr #328: improve multi-camera calibration results?

szx0112, 2019-11-22:
I am going to calibrate a four camera system using kalibr. I start with the two camera setup. With camera 0 looks straight at front, and camera 1 located 10cm above camera 0 with around 20 degrees downward. So these two cameras have significant overlap. As fig below.

[image]

I collect the images from both cameras simultaneously. The camera is not hardware sync, so I just hold the target and be still, and take the snapshots from both cameras with 3 seconds interval. I believe there is nearly no relative motions between the images. I use the april 6x6 target printed on a A4 with changed dimension in the yaml file. 

I follow the kalibr multi-camera tutorial with camera model as pinhole. However, the calibration result is not correct. See below.

[image]

The camera projection is not correct since the cameras should be the same in the y axis.

[image]

[image]

The projection error is really large compared to the demo data. I tried to remove some images with targets not fully observed by both camera. There is a slight improvement on the cam0, but the overall calibration is not correct (same as above)

[image]
 
I am not sure how could further improve this result. Any suggestion is appreciated.

NikolausDemmel, 2019-11-22:
Just two thoughts:

1. The pose looks ok to me. You have z forward, y down, x left. Looks like it's rotated around x, like it should be from your description.

2. For reprojection error, maybe model you use doesn't fit the lens so well. Could also be the calibration pattern not being perfectly flat.

szx0112, 2019-11-22:
@NikolausDemmel  
Thanks for the reply. It looks strange since there should be 0 translation between the cameras in the horizontal direction (left or right) while we could see the deviations in the pic as well as the transformation matrix

[-0.012497894441088397, -0.9161614582010943, -0.4006145093994918, **0.16430891573911857**] [0.950932082573727, 0.11297823433205689, -0.2880348813932384, **0.09965982792655875**]
[0.3091471768697645, -0.38455701927551256, 0.8697953333741215, **0.028179294748705556**]
[0.0, 0.0, 0.0, 1.0]

The april pattern is attached on a white board, so should be flat. 

I just notice maybe the images are undistorted already, maybe I should skip the intrinsic parameters estimation and correction. I have tried the 'pinhole-none' but looks like there is no such model, thus how can I skip the intrinsic parameters estimation and just do the extrinsic calibration?

NikolausDemmel, 2019-11-29:
> horizontal direction (left or right) while we could see the deviations in the pic as well as the transformation matrix

It's hard to see for me in the 3d image, due to perspective. According to your transformation matrix, there is 16cm offset in x, which you say should be 0. There is also 2.8 cm offset in z seems not too far off. And the y offset is about right with close to 10cm.

> I just notice maybe the images are undistorted already, maybe I should skip the intrinsic parameters estimation and correction. I have tried the 'pinhole-none' but looks like there is no such model, thus how can I skip the intrinsic parameters estimation and just do the extrinsic calibration?

Can you try with the original distorted images. I don't know if you can fix the intrinsics in Kalibr, but I don't think so. I would have expected models like `pinhole-equi` to degrade to pinhole model for undistorted images, i.e. all distortion parameters should then be estimated to be close to 0.

You can also try to calibrate each camera separately to find the right model. Then it's just for intrinsics, but unless you fix this, the extrinsics with multpiple cams will also not look good.

szx0112, 2019-12-12:
it makes sense. Thanks for the reply.

closed 2019-12-12 by szx0112

ShrutheeshIR, 2021-01-01:
Hello. What coordinate system does kalibr use? Is it the same as OpenCV? (x to the right, y down and z forward)
Also, for multicamera calibration, does baseline T_1_0 indicate pose of camera 1 with respect to camera 0? Or the other way around. 
In my case, cam1 is to the right (rotated to the right as well), and slightly behind cam0, here is my translation : ```[-0.17162455  0.02857572 -0.04155342]```.  Could you please explain this transformation?

Thanks