ethz-asl/kalibr #330: Is support Ubuntu 18.04 + ROS Melodic + Python3 ？

weijiang1994, 2019-11-28:
Is support Ubuntu 18.04 + ROS Melodic + Python3 ？or just support 14.04 and 16.04? thx.

gaozhiye, 2019-12-06:
> Is support Ubuntu 18.04 + ROS Melodic + Python3 ？or just support 14.04 and 16.04? thx.
This question puzzled me for 5 days,I installed it on  ubuntu1404 indigo ,but I can not got the desired result.

weijiang1994, 2019-12-06:
看你的ID应该是中国人, 我同事在16.04上面安装成功,最后也成功运行了。你可以试试16.04

jefft255, 2019-12-23:
I got it to work without too many headaches on Ubuntu 18.04 + Melodic by installing from source. Getting the Python 2 dependencies was not 100% straightforward.

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

weijiang1994, 2020-01-13:
> execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341
> 
> Thanks very much!

we acquisition two hours of imu data ,then input data to matlab script to get the imu parameters.

SasaKuruppuarachchi, 2021-09-29:
It works. 
`sudo pip install python-igraph --upgrade` didn't work but `sudo apt-get install python-igraph` worked

goldbattle, 2022-04-22:
Master should support 18.04.

closed 2022-04-22 by goldbattle