ethz-asl/kalibr #332: Cameras are not connected through mutual observations

ns15417, 2019-12-07:
when running kalibr_calibrate_cameras i encountered a problem which is listed below: 
`timestamp 	cam0 	cam1 	
1575683268.01 	84 	-
1575683268.06 	- 	56
1575683268.27 	108 	-
1575683268.32 	- 	71
1575683268.81 	- 	72
1575683269.02 	52 	-
1575683269.07 	- 	71
1575683269.26 	103 	-
1575683269.32 	- 	123
1575683269.51 	108 	-
1575683269.56 	- 	101
1575683269.77 	124 	-

**Cameras are not connected through mutual observations**, please check the dataset. Maybe adjust the approx. sync. tolerance.
Traceback (most recent call last):
  File "/home/shinan/Project/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/shinan/Project/kalibr_workspace/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 447, in <module>
    main()
  File "/home/shinan/Project/kalibr_workspace/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 204, in main
    graph.plotGraph()
  File "/home/shinan/Project/kalibr_workspace/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py", line 308, in plotGraph
    edge_label=self.G.es["weight"],
KeyError: 'Attribute does not exist'
`
above is part of the log, I am not sure what is the exact reason of this bug, but according to the bold log, it is said that the cameras are not synchronous. i guess the reason must be that i set my camera frame rate to 4HZ  which leads to the two cameras cannot get images at the same time. I think the timestamp also tels the same thing.
I solved this by changing the frame rate of both cameras to 20HZ.
Just list this here, if anyone have other explanation please tell me.

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341

Thanks very much!

AbnerCSZ, 2020-05-20:
Comment this code on line 201 of  `src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras` to solve your problem:
```
    if not graph.isGraphConnected():
        obsdb.printTable()
        print "Cameras are not connected through mutual observations, please check the dataset. Maybe adjust the approx. sync. tolerance."
        graph.plotGraph()
        sys.exit(-1)
```

huonggiangdhtn, 2022-04-27:
@ns15417 thank you, I had the same propblem, and I changed to 20Hz, it is ok. But I got error at plotGraph

closed 2022-08-18 by goldbattle