ethz-asl/kalibr #333: calibration target  file in PDF format

gaozhiye, 2019-12-07:
Hi !
I try to generate the calibration board .When I run the command line ,
          **kalibr_create_target_pdf --type apriltag --nx 6 --ny 6 --tsize 0.088 --tspace 0.3**
the reply is , command not found .
Who can  give me a calibration target  file in PDF format ?
Thank you very much !

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341 

Thanks very much!

goldbattle, 2022-06-02:
Please try on the latest master since building has been fixed. Make sure to source your ROS workspace.

closed 2022-06-02 by goldbattle

YonatanSimson, 2022-08-30:
Try this link: https://drive.google.com/file/d/1DqKWgePodCpAKJCd_Bz-hfiEQOSnn_k0/view?usp=sharing

Hezhexi2002, 2023-02-01:
> Please try on the latest master since building has been fixed. Make sure to source your ROS workspace.

Hi,I use the [Dockerfile_ros1_20_04](https://github.com/ethz-asl/kalibr/blob/master/Dockerfile_ros1_20_04) to install kalibr and also get this error too:
[image]
so is this means I have to build the kalibr from source in order to run the `kalibr_create_target_pdf` command？