ethz-asl/kalibr #334: how to use the Imu_calibration results

vigneshsara, 2019-12-13:
Hello, after I used kalibr_calibrate_imu_camera with scale-misalignment option, I got following result.
imu0:
  T_i_b:
  - [1.0, 0.0, 0.0, 0.0]
  - [0.0, 1.0, 0.0, 0.0]
  - [0.0, 0.0, 1.0, 0.0]
  - [0.0, 0.0, 0.0, 1.0]
  accelerometer_noise_density: 0.00668
  accelerometer_random_walk: 0.000917
  accelerometers:
    M:
    - [1.0075384850085152, 0.0, 0.0]
    - [0.017205839485902097, 1.0961938612921196, 0.0]
    - [-0.07519411884856604, 0.11078387285861767, 1.0208877441714266]
  gyroscope_noise_density: 0.00203
  gyroscope_random_walk: 0.000271
  gyroscopes:
    A:
    - [0.00033628818679622813, 0.00033989727403612715, 0.0023460575365993577]
    - [-0.00021309674291592032, -0.00026022043328967305, -0.0003105405149146512]
    - [-2.070095610881228e-06, 0.0003658263740224834, 0.0015392312208048244]
    C_gyro_i:
    - [0.997225320211812, -0.07420530203968216, 0.005935813141207369]
    - [0.07418323837859316, 0.9972371766621109, 0.003854948332391877]
    - [-0.006205471143484005, -0.00340391424394451, 0.9999749524341627]
    M:
    - [0.40473723961271335, 0.0, 0.0]
    - [0.0012193641786323738, 0.46467919471887387, 0.0]
    - [0.007563932319210696, -0.005224639240317598, 0.5023767098998477]
  model: scale-misalignment
  rostopic: /taraxl/imu/data_raw
  time_offset: 0.0
  update_rate: 26.0
And i had use this formal c=M*(R-B) to convert the raw to correct data. 
m=missaliment data ,R=raw data from camera(uncalibrated data) ,B= random_walk 
i got work result like (0.0045,11.4598,0.0087) instants  of (0.000,9.887,0.799).
its this right way to calculate the results ,or use any other formulas are avalible ???

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341 

Thanks very much!

mhyoosefian, 2021-12-03:
Hi,

See [this](https://github.com/ethz-asl/kalibr/issues/189#issuecomment-978890978) comment.

Regards,
Mohammad