ethz-asl/kalibr #336: found incorrect value on imu extrinsic

vigneshsara, 2019-12-20:
i had attached my result below,baseline my camera 60mm its give correctly but compare with imu distance between cam0 to imu is 15mm and imu to camera1 is 45 but kalibr result are incorrect. any idea for this results???     
Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[-0.09063329  0.99570995  0.01863606  0.0004547 ]
 [-0.99546944 -0.09003956 -0.03055279  0.00031908]
 [-0.02874373 -0.02132073  0.99935941  0.00004048]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[-0.09063329 -0.99546944 -0.02874373  0.00036001]
 [ 0.99570995 -0.09003956 -0.02132073 -0.00042316]
 [ 0.01863606 -0.03055279  0.99935941 -0.00003918]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
0.00495656047192

Transformation (cam1):
-----------------------
T_ci:  (imu0 to cam1): 
[[-0.09016874  0.99573806  0.01937282 -0.05960656]
 [-0.99562357 -0.08964452 -0.02641142  0.00019255]
 [-0.02456219 -0.02166952  0.99946342  0.00035414]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam1 to imu0): 
[[-0.09016874 -0.99562357 -0.02456219 -0.00517424]
 [ 0.99573806 -0.08964452 -0.02166952  0.05937746]
 [ 0.01937282 -0.02641142  0.99946342  0.00080589]
 [ 0.          0.          0.          1.        ]]

timeshift cam1 to imu0: [s] (t_imu = t_cam + shift)
0.00480246314822

Baselines:
----------
Baseline (cam0 to cam1): 
[[ 0.99999962 -0.00048748  0.00072233 -0.06006114]
 [ 0.00048449  0.99999133  0.00413473 -0.00012691]
 [-0.00072434 -0.00413438  0.99999119  0.00031531]
 [ 0.          0.          0.          1.        ]]
baseline norm:  0.0600621048591 [m]

Gravity vector in target coords: [m/s^2]
[-0.07590761  0.09258016 -9.80581918]

kannidekan, 2020-01-11:
Man, i looked much questions of you. If i can help you, i will.

But i am a newer, haha.
could you answer me a little question please #341 
There seems no much people to answer, so could you help me please.

goldbattle, 2022-04-27:
Please provide more details on what the issue is.

closed 2022-04-27 by goldbattle