ethz-asl/kalibr #337: Two Pointgrey Camera Calibration

meihouking, 2019-12-25:
[image]
[image]
I try to use Kalibr to calibrate two pointgrey Camera. These images almost have no distortion. The distance of two cameras is x=6.4~6.7cm , Y= Close to 0, Z= Close to 0. but the result is wrong.
 Cam resolution: [1920, 1200], Cam model : pinhole-fov
I dont know what problem cause the result wrong?

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341 

Thanks very much!

goldbattle, 2022-06-02:
FOV is normally used for fisheye cameras which you don't seem to have. I recommend you try the other distortion models. Please provide a bag where you have issues running / expand the exact problem you have, thanks!

closed 2022-06-16 by goldbattle