ethz-asl/kalibr #340: A question of addCameraChainErrorTerms

EmiyaEstelle, 2020-01-03:
Hi，recently I have been reading the codes of `kalibr_calibrate_imu_camera`. When I read `addCameraChainErrorTerms()`, which the function is to add the reprojection error terms for all cameras, in the `IccSensors.py`, I have felt uncertainty. Here are the  codes 
```
    #add the reprojection error terms for all cameras in the chain
    def addCameraChainErrorTerms(self, problem, poseSplineDv, blakeZissermanDf=-1, timeOffsetPadding=0.0):
        
        #add the induviduak error terms for all cameras
        for camNr, cam in enumerate(self.camList):
            #add error terms for the first chain element
            if camNr == 0:
                #initialize the chain with first camerea ( imu to cam0)
                T_chain = cam.T_c_b_Dv.toExpression()
            else:
                T_chain = cam.T_c_b_Dv.toExpression() * T_chain
            
            #from imu coords to camerea N coords (as DVs)
            T_cN_b = T_chain
            
            #add the error terms
            cam.addCameraErrorTerms( problem, poseSplineDv, T_cN_b, blakeZissermanDf, timeOffsetPadding )
```
I have a question about T_cN_b. In its notes, T_cN_b is from imu coordinate systems (b) to the N-th camera coordinate systems(cN). There, we assume only two cameras and one imu. When the camera number is 0 (the first camera), `T_chain = T_c1_b`. Then, the camNr = 1 and `T_c2_b = T_c2_b * T_chain`, where T_chain is T_c1_b. 
I have no idea about the equation `T_chain = cam.T_c_b_Dv.toExpression() * T_chain`. In my case above, the matrix T from imu to second camera is equal to T_c2_b * T_c1_b? In my opinion. the T_c2_b should be _T_c2_c1? Thank you!

kannidekan, 2020-01-11:
execuse me, do you know how could i get the IMU YAML file of the requirements in the camera-imu calibration? #341 

Thanks very much!

EmiyaEstelle, 2020-01-13:
@kannidekan 
Hi. The IMU YAML should be got by the third party tool. I used it [()](https://github.com/gaowenliang/imu_utils) and you can try it.