ethz-asl/kalibr #341: [Help] How to create IMU configuration YAML of the requirements in the Camera IMU calibration.

kannidekan, 2020-01-11:
Thanks your solid work for the camera-imu calibration very much.

I'm a newer to this field. Can you or others answer me a little question?

There is a need of IMU YAML file at [camera-imu calibration](https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration).

> --imu imu.yaml
> contains the IMU statistics and the IMU's topic (see YAML formats)

Then, I went to have a look at [IMU intrinsic calibration](https://github.com/ethz-asl/kalibr/wiki/Multi-IMU-and-IMU-intrinsic-calibration)

I have read all of these two files. Beacuse the second one is the extended function with respect to the first one as you mentioned.
But I haven't seen any command about how to calibrate a single IMU!!!

> 3) Running the calibration
> Please see here for the required options governing camera configuration as well as ROS-bag and target selection.

Here, does it mean calibrating the imu by this command:
`kalibr_calibrate_imu_camera --bag [filename.bag] --cam [camchain.yaml]  --target [target.yaml]`

If not, how i can get the imu.yaml?

I really need your help. There is no much time for my study work. 
Thanks for your time to look anyway, best wish!

zxawangxiaomu, 2020-01-13:
you can get this yaml file follow this page https://github.com/ethz-asl/kalibr/wiki/Multi-IMU-and-IMU-intrinsic-calibration

kannidekan, 2020-01-13:
> you can get this yaml file follow this page https://github.com/ethz-asl/kalibr/wiki/Multi-IMU-and-IMU-intrinsic-calibration

Thanks your replay very much. @zxawangxiaomu 

But what's the commond i shold run? I have read this page. It is said the extented function to the original `kalibr_calibrate_imu_camera` tool. 

Should i run the command `kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45` as the same as the calibration of camera and IMU?

But if run as this command, the the requirement of the option `--imu` is what i calibrate. It seems confusing.

kannidekan, 2020-01-13:
@zxawangxiaomu 

I also find the [`ZED-M` official calibration](https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-) by the kalibr.

**The IMU YMAL is directly given**. So, is the file just a initial setting and will the tool to fix the real params?

zxawangxiaomu, 2020-01-13:
do you mean the IMU noise? it can obtain from "Allan standard deviation" just the page said https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model, or use the open source https://github.com/gaowenliang/imu_utils

vigneshsara, 2020-01-13:
@kannidekan HI ,i had used the imu_utils for creating the imu.ymal file,but i think  it's does not gave the correct result.
Imu noise value are available in datasheet also. 
thank you

kannidekan, 2020-01-14:
Both of you are warmhearted.  @zxawangxiaomu @vigneshsara 
Thanks very much, best wishes for your work and study!

closed 2020-01-16 by kannidekan