ethz-asl/kalibr #342: Enable multi-thread during optimization

zhixy, 2020-01-15:
I notice that the number of threads is set according to cpu core[here](https://github.com/ethz-asl/kalibr/blob/14e47ea0edb7a42efa03533855e533f9e6ad7bb8/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py#L158), however when I really calibrate camera-imu, only one thread is been used as I make sure that with system monitor.

kongdaqing, 2020-03-12:
I have same problem! how to solve this!

JACKLiuDay, 2022-01-13:
I have the same problem too, waiting so long time for the result of calibration

JACKLiuDay, 2022-02-14:
> I have same problem too!

Hi, you can build your kalibr with release. Only release version can run fast.

DavidNY123, 2022-02-14:
> > I have same problem too!
> 
> Hi, you can build your kalibr with release. Only release version can run fast.

Hello,thank u for your reply! But I still have this problem.Only one thread is enable

JACKLiuDay, 2022-02-15:
> 

catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release
Try this command. I have the same problem before, and I build the kalibr with this command. It worked for me.

JACKLiuDay, 2022-02-15:
> > > I have same problem too!
> > 
> > 
> > Hi, you can build your kalibr with release. Only release version can run fast.
> 
> Hello,thank u for your reply! But I still have this problem.Only one thread is enable

catkin build --cmake-args -DCMAKE_BUILD_TYPE=Release
You can try this command and it worked for me.

goldbattle, 2022-05-04:
This is actually due to the backend solver `BlockCholeskyLinearSystemSolver` not supporting multiple threads. It would be nice to do what is described in it:
https://github.com/ethz-asl/kalibr/blob/e9453e0734abec221d4335b1d6fb3202b1c9ff2b/aslam_optimizer/aslam_backend/src/BlockCholeskyLinearSystemSolver.cpp#L58-L72