ethz-asl/kalibr #343: Multi-camera calibration: Query regarding image resolution and using intrinsic information as an input to the Kalibr tool

adityarasam, 2020-01-15:
Hi
I am trying to generate a Depth image using a RGB camera and a point cloud from 3D-depth sensor camera. In this case, it is quite essential to have correct extrinsic information between two cameras to register/overlay the data from both cameras accurately. I am using Kalibr multi-camera calibration tool to generate that extrinsic information where I feed gray scale image from RGB camera and IR images from from 3D-depth sensor. However, I see that Kalibr also generates the intrinsic information for all the cameras and I assume that it might be using this information to generate the extrinsic transformation between them. In my case, I already have the intrinsic information for both of these sensors which are factory calibrated and hence accurate. I wanted to know if there is any way where we can give the intrinsic value to the Kalibr multi-camera calibration tool as an input, so that it can use that information to compute the extrinsic between the camera?

Secondly, the IR images from the 3D-depth sensor are 224x172 in resolution and RGB camera has 640 x 480 resolution. Would the difference in resolution affect the accuracy of the result from multi-camera calibration tool? Also what should be a good re-projection error for cameras with such resolution?

ruoyiyi, 2020-08-19:
Hi @adityarasam , I have the same problem as you.  Did you find a solution to input the camera intrinsic parameters to the Kalivbr multi-camera calibration tool?

colorDogs, 2020-08-28:
I have the same problem

ssssjiang, 2021-08-31:
I have the same problem

goldbattle, 2022-05-04:
This is being tracked in https://github.com/ethz-asl/kalibr/issues/23 and currently reading and fixing intrinsics is not supported.

closed 2022-05-04 by goldbattle