ethz-asl/kalibr #344: No corners could be extracted for low resolution images( confidence image )

arenas7307979, 2020-01-17:
I am trying to calibrate a tof camera（TI TINTIN)

the problem is that for every dataset  I always receive the error
[FATAL] [1572255263.563560]: No corners could be extracted for ...

the image below.
[image]

sebasantibanez, 2020-05-13:
Hi @arenas7307979, I'm in the same boat.  Did you manage to solve it?

arenas7307979, 2020-05-19:
I think it has something problem from the sharpness of the april_calibration image, which leads to the lack of feature points,  maybe u can using general chessboard to calibration

goldbattle, 2022-05-04:
I think tuning the camera parameters or doing some preprocessing will be necessary to have enough gradient for detection. If you do find a solution, feel free to post for others to see. Additionally, posting a bag would help others debug.

tejalbarnwal, 2022-06-06:
hey, it might be possible that the images are mirrored. Please check for the same

closed 2023-05-24 by goldbattle