ethz-asl/kalibr #349: How to set KALIBR_MANUAL_FOCAL_LENGTH_INIT?

SubMishMar, 2020-02-14:
The camera focal length initialization fails for me for one of my cameras and I am not sure why. But I realized that I can set KALIBR_MANUAL_FOCAL_LENGTH_INIT but I don't know how. Any help?

Thanks.

SubMishMar, 2020-02-14:
I got it we have to set it in the terminal. Like:
export KALIBR_MANUAL_FOCAL_LENGTH_INIT=2000
And then again set this value when the code asks for it.

closed 2020-02-14 by SubMishMar

floriantschopp, 2020-02-15:
`KALIBR_MANUAL_FOCAL_LENGTH_INIT` is just a flag whether the option to enable manual input should be used or not (as it can be different for different cameras in the same optimization). This was done to be able to set it somewhere in your `bashrc` and never care about it again if you are not using a scriped version of kalibr as discussed here #346