ethz-asl/kalibr #354: Kalibr IMU Noise Model units problem?

maximpavliv, 2020-03-16:
I am using your wonderful tool to calibrate my eye-tracking device. 

I am intrigued by the IMU noise model as described [here](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model)  : You describe how to find the std of the noise and the random walk using the Allan standard deviation plot. 
If I understand it correctly, the Allan plot would give the continuous std and random walk of the noise, in [_(rad/s)*sqrt(s)_] for the angular velocity and in [_(m/s2)*sqrt(s)_] for the accelerations. You also give the formulas to convert it to discrete std noise and random walk values, which I would expect to have the same units as the original units of the measurements ([_m/s^2_] or [_rad/s_]). The formula you give for the std of the noise (sigmad = sigmac / sqrt(dt)) does give [_m/s^2_] for the acceleration and [_rad/s_] for the angular velocities, but the formula for the random walk discretisation (sigmad = sigmac * sqrt(dt)) would give [_m/s_] as a unit for the acceleration random walk and [_rad_] as a unit for the angular velocity random walk. Are these the real units of random walks, or is the random walk discretization formula not correct?

ToniRV, 2020-07-10:
I second this issue.

According to http://mars.cs.umn.edu/tr/reports/Trawny05b.pdf 
I think the units should be:
[image]

Maybe someone more familiar with Kalibr way of doing things can verify this.

mintar, 2021-11-26:
@maximpavliv wrote:
> I am intrigued by the IMU noise model as described [here](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model): You describe how to find the std of the noise and the random walk using the Allan [...]

The Allan plot gives the same units as the ones listed on the Kalibr wiki page that you linked to:
* "white noise" (= "angle random walk"/"velocity random walk"/"noise density"): [_(rad/s) * (1/√Hz)_] for gyros / [_(m/s²) * (1/√Hz)_] for accelerometers, which is equivalent to the units you used ([_(rad/s)*sqrt(s)_] for the angular velocity and in [_(m/s^2)*sqrt(s)_] for the accelerations).
* "random walk" (= "rate random walk"/"acceleration random walk"/"random walk"): [_(rad/s²) * (1/√Hz)_] for gyros / [_(m/s³) * (1/√Hz)_] for accelerometers

> You also give the formulas to convert it to discrete std noise and random walk values, which I would expect to have the same units as the original units of the measurements ([_m/s^2_] or [_rad/s_]). [...]

Everything correct so far.

> but the formula for the random walk discretisation (sigma{d} = sigma{c} * sqrt(dt)) would give [_m/s_] as a unit for the acceleration random walk and [_rad_] as a unit for the angular velocity random walk.

No, because as I wrote above (and as the table in the wiki page shows) the "random walk" parameter has a different unit, so the formula <img src="https://latex.codecogs.com/svg.image?\sigma_{bgd}=\sigma_b_g\sqrt{\Delta&space;t}" title="\sigma_{bgd}=\sigma_b_g\sqrt{\Delta t}" /> from the wiki page gives [_(rad/s²) * (1/√Hz) * √s = (rad/s²) * √s * √s = rad/s_] for gyros and [_(m/s³) * (1/√Hz) * √s = (m/s³) * √s * √s = m/s²_] for accelerometers.

By the way, here's a small python implementation of the formulas on the wiki page that will simulate IMU measurements based on the continuous `noise_density` and `random_walk` parameters:

```python
import numpy as np

# n: number of measurements
# dt: sampling rate [s], e.g. dt = 1e-2 when sampling at 100 Hz
def generate_signal(n, dt, noise_density, random_walk, random_state=0):
    rng = np.random.RandomState(random_state)
    white = noise_density / np.sqrt(dt) * rng.randn(n)
    walk = random_walk * np.sqrt(dt) * np.cumsum(rng.randn(n))
    return white + walk
```

@ToniRV:

First off thanks for the link to the Trawny paper, that was very helpful.

The units that you propose are equivalent to the ones in the table on the Kalibr wiki page:
* your "white noise" units already match
* for the "random walk" units: rad/s * √Hz = rad/s * √Hz * √Hz/√Hz = rad/s * Hz * (1/√Hz) = rad/s * (1/s) * (1/√Hz) = (rad/s²) * (1/√Hz), and analogously for the accelerometer.

Your units look a bit nicer (because it immediately makes it clear that you have to multiply/divide by frequency, or equivalently divide/multiply by sampling rate), but they are equivalent.

**TL;DR:** In summary, the Kalibr wiki page is correct (although it took me a very long time to verify this, because it was inconsistent with lots of other software packages, but it turned out that those were incorrect). I think this issue can be closed.