ethz-asl/kalibr #364: Cameras are not connected through mutual observations

liubamboo, 2020-05-25:
when I ran command below to calibrate realsense D435i rgb cemera, the bug always appears

command: kalibr_calibrate_cameras --target ~/project/calibration/rgb/april_6x6_A4.yaml --bag ~/disk/datasets/realsense-dataset/calibration/rgb/rgb.bag --models pinhole-radtan --topics /camera/color/image_raw --bag-from-to 20 80 --show-extraction
 
Cameras are not connected through mutual observations, please check the dataset. Maybe adjust the approx. sync. tolerance.
Traceback (most recent call last):
  File "/home/n609/software/kalibr_workspace/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/n609/software/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 447, in <module>
    main()
  File "/home/n609/software/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 204, in main
    graph.plotGraph()
  File "/home/n609/software/kalibr_workspace/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py", line 311, in plotGraph
    edge_label=self.G.es["weight"],
KeyError: 'Attribute does not exist'

I try to add --approx-sync like this:
kalibr_calibrate_cameras --target ~/project/calibration/rgb/april_6x6_A4.yaml --bag ~/disk/datasets/realsense-dataset/calibration/rgb/rgb.bag --models pinhole-radtan --topics /camera/color/image_raw --bag-from-to 20 80 --show-extraction --approx-sync 0.01

but it doesn't work.and I try to change the Hz of rgb cemera.but it doesn't work too. now I don't know how I can solve it. Anyone body know how to fix the problem? if you know , please tell me ,thanks very much!

armanhamzehlou, 2020-05-30:
I just had this issue couple of hours ago and I tried increasing approx-sync to 5. it no longer complained.

GGDali, 2020-06-17:
modify the file  aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py at line 83
#############################################################    
    #check if all cams are connected through observations
    def isGraphConnected(self):
        #check if all vertices are connected
        return self.G.adhesion()

        if self.numCams == 1:
            # Since igaph 0.8, adhesion correctly returns 0 for the non-connected one cam case.
            #   which evaluates to false later on. So we skip the check and return true in the one camera case.
            return True
        else:
            #check if all vertices are connected
            return self.G.adhesion()

    #returns the list of cam_ids that share common view with the specified cam_id
    def getCamOverlaps(self, cam_id):

GGDali, 2020-06-17:
modify the file  aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py at line 83 
`#############################################################    
    #check if all cams are connected through observations
    def isGraphConnected(self):
        #check if all vertices are connected
        return self.G.adhesion()

        if self.numCams == 1:
            # Since igaph 0.8, adhesion correctly returns 0 for the non-connected one cam case.
            #   which evaluates to false later on. So we skip the check and return true in the one camera case.
            return True
        else:
            #check if all vertices are connected
            return self.G.adhesion()

    #returns the list of cam_ids that share common view with the specified cam_id
    def getCamOverlaps(self, cam_id):`

bednarhonza, 2020-07-08:
Thanks! Actually, this should be changed in the code for everybody. Could you please do it? You have saved me a lot of time..

liaozihang, 2021-09-09:
> modify the file aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py at line 83
> `#############################################################
> #check if all cams are [...]

thx

mzahana, 2021-09-27:
@stereolabs Can you please adapt these changes in your Docker image?

alwynmathew, 2022-03-22:
@mzahana thank you for [your docker image](https://hub.docker.com/r/mzahana/kalibr) for the single image kalibr tool.

mzahana, 2022-03-22:
@alwynmathew you are welcome !

closed 2022-04-22 by goldbattle

Pplll-pppl, 2026-02-13:
@mzahana 

Hi, have you adapted these changes you mentioned in your Kalibr docker image?