ethz-asl/kalibr #367: Manually give initial estimates of calibration parameters.

johnny04thomas, 2020-06-23:
I would like to manually input initial estimates of calibration parameters (intrinsic and distortion) from previously calibrated values in Multiple camera calibration. Please help me identify the file and line to modify for fisheye models: omni-radtan and ds. Thanking in advance.

haoyuanz13, 2021-05-24:
any update here?

johnny04thomas, 2021-05-24:
Yes, I managed to do that. Sorry for not updating about it before. In kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras once the initial individual intrinsic parameters are evaluated, you could set your calibration parameters as the initial guesses for the intrinsics inside the main while loop. 
    ```
    cameraList[0].geometry.projection().setParameters(np.array(projection_parameters))
    cameraList[0].geometry.projection().distortion().setParameters(np.array(distortion_parameters))
```
After that the intrinsics initialized are used as known parameters in the optimization by deactivating their role as the design variable. The following modification is required to be made inside **CameraCalibrator.py** file in **kalibr_camera_calibration** folder.
    `177   camera.setDvActiveStatus(False, False, False)`
I hope this helps.

haoyuanz13, 2021-05-31:
Good to KNOW that!

I have tried your instructions and it works which fixes the initial intrinsic params during the optimization, thanks for your reply!

johnny04thomas, 2021-06-04:
You're Welcome!

ishipachev, 2022-01-11:
@johnny04thomas 
Thank you. It helped me too.

foemre, 2023-06-09:
Could be worthwhile to consider adding the option for manual entry of initial guesses.