ethz-asl/kalibr #368: Error with boost dependency

sebastian-ruiz, 2020-06-29:
Following the installation guide, by first cloning the repository to `~/catkin_ws/src`, I get the following error when running:

```
$ catkin_make -DCMAKE_BUILD_TYPE=Release
```

The error is:
```
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Could not find a package configuration file provided by "boost_python3"
  (requested version 1.71.0) with any of the following names:

    boost_python3Config.cmake
    boost_python3-config.cmake

  Add the installation prefix of "boost_python3" to CMAKE_PREFIX_PATH or set
  "boost_python3_DIR" to a directory containing one of the above files.  If
  "boost_python3" provides a separate development package or SDK, be sure it
  has been installed.
```

neutrinoyu, 2020-06-30:
I also get the same error .
Inspired by this [https://github.com/ros-melodic-arch/ros-melodic-cv-bridge/pull/4](url)

/kalibr/Schweizer-Messer/numpy_eigen/cmake/add_python_export_library.cmake:89
change
`    list(APPEND BOOST_COMPONENTS python3)`
to
`    list(APPEND BOOST_COMPONENTS python)`
Hope it helps.

jasonbeach, 2021-03-01:
I just bumped into this as well on noetic w/ Ubuntu 20.04.  In my case, I had to edit  `vision_opencv/cv_bridge/CMakeLists.txt` as:

```
if(NOT ANDROID)
  find_package(PythonLibs)
  #  if(PYTHONLIBS_VERSION_STRING VERSION_LESS 3)
    find_package(Boost REQUIRED python)
    #else()
    #find_package(Boost REQUIRED python3)
    #endif()
else()
find_package(Boost REQUIRED)
endif()
```

i.e. basically comment out the test for checking for python2.

jasonbeach, 2021-03-01:
> I just bumped into this as well on noetic w/ Ubuntu 20.04. In my case, I had to edit `vision_opencv/cv_bridge/CMakeLists.txt` as:
>
> ```
> if(NOT ANDROID)
>   find_package(PythonLibs)
>   # [...]

actually it turns out vision_opencv has a `noetic` branch I just had to switch to

lhj9, 2022-01-10:
> 

Yeah, I use your method and it works, thanks.

goldbattle, 2022-04-22:
Hi please try PR https://github.com/ethz-asl/kalibr/pull/515 to see if that solves your issue.

closed 2022-05-03 by goldbattle