ethz-asl/kalibr #370: kalibr_calibrate_cameras NameError: global name 'errList' is not defined

zsk194189650, 2020-07-05:
Hello,when use kalibr_calibrate_cameras, there is an err like this: 

sycon@Sycon:~/kalibr_ws$ kalibr_calibrate_cameras --bag fisheye_calibra.bag --topics /left --models pinhole-radtan --target april_3x2_5cm_A1.yaml
importing libraries
Initializing cam0:
	Camera model:	  pinhole-radtan
	Dataset:          fisheye_calibra.bag
	Topic:            /left
	Number of images: 191
Traceback (most recent call last):
  File "/home/sycon/kalibr_ws/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/sycon/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 447, in <module>
    main()
  File "/home/sycon/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 172, in main
    cam = kcc.CameraGeometry(cameraModel, targetConfig, dataset, verbose=(parsed.verbose or parsed.showextraction))
  File "/home/sycon/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraCalibrator.py", line 47, in __init__
    self.ctarget = TargetDetector(targetConfig, self.geometry, showCorners=verbose, showReproj=verbose)
  File "/home/sycon/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraCalibrator.py", line 80, in __init__
    targetParams = targetConfig.getTargetParams()
  File "/home/sycon/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 186, in func
    return f(*args, **kwargs)
  File "/home/sycon/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 610, in getTargetParams
    errList.append("invalid tagRows (int>=3)")
NameError: global name 'errList' is not defined

Can anyone tell me why it happened and how to solve it.
Thanks!

zsk194189650, 2020-07-08:
The size of roapriltag must be greater than 3*3.

closed 2020-07-08 by zsk194189650

heisetiantang, 2022-11-10:
[image]
[image]
[image]
[image]

我与作者出现了相同的问题，我的标定板是8×6的，是大于3×3的，为什么任然出现上述问题，是否可能因为april的命名方式，必须加上a*b和每一格的大小，

I have the same problem with the author. My calibration plate is 8×6, which is larger than 3×3. Why does the above problem still occur?

wbl17792615320, 2023-05-09:
I am not very understand, but I am glad that I have solved the problem, my Ubuntu system is 18.04, should be related to the installation problems, must adapt to their own system, if you are 18.04 system, can follow this tutorial to do :https://blog.csdn.net/ndwonduosk_/article/details/124718186?spm=1001.2014.3001.5506
18.04Do not use the catkin _ make command