ethz-asl/kalibr #374: multi camera calibrate throws numpy error

LXYYY, 2020-07-30:
Hi, i am tring to use kalibr to calibrate multi cameras, but I got the following error:
```
initializing initial guesses
	 initializing camera pair (0,2)...  
/home/lxy/.local/lib/python2.7/site-packages/numpy/lib/function_base.py:4011: RuntimeWarning: Invalid value encountered in median for 3 results
  r = func(a, **kwargs)
Traceback (most recent call last):
  File "/home/lxy/Workspace/mrslam/kalibr_ws/devel/bin/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/lxy/Workspace/mrslam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 449, in <module>
    main()
  File "/home/lxy/Workspace/mrslam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 218, in main
    baseline_guesses = graph.getInitialGuesses(cameraList)
  File "/home/lxy/Workspace/mrslam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/MulticamGraph.py", line 178, in getInitialGuesses
    distortionActive=False)
  File "/home/lxy/Workspace/mrslam/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraIntializers.py", line 32, in stereoCalibrate
    r_median = np.median(np.asmatrix(r), axis=0).flatten().T
  File "/home/lxy/.local/lib/python2.7/site-packages/numpy/lib/function_base.py", line 4097, in median
    overwrite_input=overwrite_input)
  File "/home/lxy/.local/lib/python2.7/site-packages/numpy/lib/function_base.py", line 4011, in _ureduce
    r = func(a, **kwargs)
  File "/home/lxy/.local/lib/python2.7/site-packages/numpy/lib/function_base.py", line 4150, in _median
    return np.lib.utils._median_nancheck(part, rout, axis, out)
  File "/home/lxy/.local/lib/python2.7/site-packages/numpy/lib/utils.py", line 1159, in _median_nancheck
    result[n] = np.nan
ValueError: non-broadcastable output operand with shape (1,3) doesn't match the broadcast shape (3,3)
```
To me, it looks like maybe a numpy version issue? anyone met the same issue and have some solution?

LXYYY, 2020-07-30:
And following some stackoverflow question, I added this `np.seterr(invalid='ignore')` into main, but still same error

goldbattle, 2022-04-22:
Can you provide a ROS bag where this is an issue? Thanks!

goldbattle, 2023-02-13:
This is likely caused by not being able to recover the pose of the target via PnP. Post a bag if you are able to and reopen then.

closed 2023-02-13 by goldbattle