ethz-asl/kalibr #375: Bagcreator No IMU Information

sooroosh, 2020-08-02:
Hi,
I am trying to Calibrate a Ricoh Theta V camera with two 190 degree back to back camera sensors. I have recorded the calibration images with the apriltags and put the images coming from each one of the cameras in the correct folder structure mentioned in the tutorial. However, I have no IMU information. 

In the tutorial it says that having a csv file for the IMU is optional. However, when I run the bagcreator script from the CDE package I get the following error:
  File "<string>", line 73, in loadImageToRosMsg
ValueError: invalid literal for int() with base 10: ''"

Looking at the other issues I believe it has something to do with the missing CSV file. I wonder if there is a way to get this to work without any IMU information ?

sooroosh, 2020-08-03:
Turns out the problem is with images names ! Images names should represent the timestep and should at least be 10**9 long (1 second in nanoseconds) !
Therefore if your image numbering starts from 0 , add an offset of 10**9 (or greater) to all image names.

closed 2020-08-03 by sooroosh

DepinYang, 2020-08-19:
@sooroosh Hi, I am also calibrating RICOH Theta V. Could you tell me which camera model did you choose? `omni-radtan` or `ds-none` or something else?