ethz-asl/kalibr #379: Omnidirectional camera model

Eashwar93, 2020-08-19:
I have used omnidirectional camera model to calibrate two of my mono cams using multicamera calibration and used the image_undistort package to get the corrected images.

The camera resolution is 1280x720 but after calibration and undistortion the resolution is 2705 x 968. 

I'm running object tracking in 3D space in one camera 1 and object re-ID in camera 2. The object locations detected in camera 1 has to be projected onto the 2D image coordinates of camera 2 for this purpose. 

When I take a look at the intrinsics vector from Kalibr for an omnidirectional camera model there are 5 parameters xi,fu,fv,pu,pv. 

I'm aware of constructing a projection matrix with pinhole camera model. I'm not sure as to how I need to interpret the parameters of omnidirectional camera model in the projection matrix as there is an additional parameter xi.

If anyone can tell me as to how I should construct the projection matrix from the intrinsics vector presented here, that would be of great help. I'm aware that I need to make use of the transform between the camera coordinates but not sure how the intrinsic parameters need to be used to construct the projection matrix.

NikolausDemmel, 2020-08-20:
After undistortion, the images should follow the pinhole model. I'm not sure how image_undistort works. If you specify the "output" intrinsics manually, you should know them. If it determines them automatically, you need to see where you can read them. Maybe they are printed out somewhere by image_undistort.

If you want to work on the distorted images instead, the non-linear projections like the omni model cannot be expressed as a "projection matrix". If you are interested how the `xi` parameter is used in the projection equations, you can check the concise summary in [this paper](https://arxiv.org/pdf/1807.08957.pdf). There it's called "Unified Camera Model". Beware that AFAIK Kalibr additionally adds radial tangential distortion afterwards, when you select "omni-radtan" (this is not described in the linked paper).

Eashwar93, 2020-09-11:
@NikolausDemmel thanks a lot for your response. I found it in one of the topics image_distort was publishing :)

goldbattle, 2022-04-27:
Wiki has been updated with references based on @NikolausDemmel details: https://github.com/ethz-asl/kalibr/wiki/Supported-models

closed 2022-04-27 by goldbattle