ethz-asl/kalibr #381: Trting to use bags created by kalibr_bagcreater : deleted by other programs

amy-tabb, 2020-08-28:
Hi, I am trying to create a dataset created with non-ROS image acquisition tools and I am using the CDE version,

so I create directories according to [bagcreater](https://github.com/ethz-asl/kalibr/wiki/bag-format#bagcreater), 

```bash
./kalibr_bagcreater --folder /home/atabb/git/kalibr-testing-aug/kalibr-cde/calico/data/ --output-bag /home/atabb/git/kalibr-testing-aug/kalibr-cde/calico.bag
importing libraries
```
Seems to work.  No errors.

Then, I try to calibrate the two cameras using `kalibr_create_cameras`:

```bash
./kalibr_calibrate_cameras --target calico/april_10x6.yaml --bag calico.bag --models pinhole-radtan pinhole-radtan --topics /cam0/image_raw /cam1/image_raw
importing libraries
Initializing cam0:
	Camera model:	  pinhole-radtan
	Dataset:          /home/atabb/git/kalibr-testing-aug/kalibr-cde/calico.bag
	Topic:            /cam0/image_raw
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 444, in <module>
  File "<string>", line 165, in main
  File "<string>", line 29, in initBagDataset
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 29, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 159, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 890, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 903, in _open_read
    self._file     = open(f, 'rb')
IOError: [Errno 2] No such file or directory: '/home/atabb/git/kalibr-testing-aug/kalibr-cde/calico.bag'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_cameras", line 5, in <module>
    exec(fh.read())
  File "<string>", line 444, in <module>
  File "<string>", line 165, in main
  File "<string>", line 29, in initBagDataset
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 29, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 159, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 890, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 903, in _open_read
    self._file     = open(f, 'rb')
IOError: [Errno 2] No such file or directory: '/home/atabb/git/kalibr-testing-aug/kalibr-cde/calico.bag'
```

The bag file is deleted.  I have installed python3-apport and python-apport with no change.

I recreate the bag file, and test the extractor, which worked fine with the sample file.

```bash
./kalibr_bagextractor --image-topics /cam0/image_raw /cam1/image_raw  --output-folder output --bag calico.bag 
importing libraries

Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagextractor", line 5, in <module>
    exec(fh.read())
  File "<string>", line 46, in <module>
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 29, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 159, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 890, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 903, in _open_read
    self._file     = open(f, 'rb')
IOError: [Errno 2] No such file or directory: '/home/atabb/git/kalibr-testing-aug/kalibr-cde/calico.bag'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_bagextractor", line 5, in <module>
    exec(fh.read())
  File "<string>", line 46, in <module>
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImageDatasetReader.py", line 29, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 159, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 890, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rosbag/bag.py", line 903, in _open_read
    self._file     = open(f, 'rb')
IOError: [Errno 2] No such file or directory: '/home/atabb/git/kalibr-testing-aug/kalibr-cde/calico.bag'
```

again, the bag file is deleted and it does not work. 

Any ideas on how to resolve this issue?  I would prefer not to build from source since I am not familiar with ROS and the version is so old.

LarryDong, 2021-08-05:
I met the same problem, and I found the solution from [this issue](https://github.com/ethz-asl/kalibr/issues/277)

The reason is, the `kalibr_calibrate_camera` scripts delete some old files in its folder and you put the bag to exact the script's folder. Just move the rosbag to another folder.

Hope this can help others.

goldbattle, 2022-04-22:
Will track this fix via #277

closed 2022-04-22 by goldbattle