ethz-asl/kalibr #388: How do I get the IMU calibration file?

SparkyTD, 2020-10-09:
I have a device with two IR cameras in stereo configuration and an IMU. I want to use kalibir to calibrate them and to get the final camchain-imucam.yaml file that I can use with other projects, such as [msckf_vio](https://github.com/KumarRobotics/msckf_vio).

I can create a rosbag with the two camera topics and the imu data using [this](https://github.com/ethz-asl/kalibr/wiki/bag-format) tutorial, but I don't know what's the next step. Do I have to calibrate my IMU separately first, and then the cameras? All the tutorials on the wiki seem to rely on some sort of IMU_YAMLS and IMU_MODELS files, and I don't know where to get them from.

What is the exact order of steps that I have to take to get the final calibration file?

Neel125, 2020-11-03:
I used Pixhawk to get imu data. But don't know how to get the IMU-noise model. 
Please help me.

goldbattle, 2022-04-27:
See https://github.com/ethz-asl/kalibr/issues/76. You need to record a long statationary bag or look at the IMU datasheet for details. See the wiki https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model