ethz-asl/kalibr #393: kalibr_calibrate_imu_camera  failed for imu and camera

vishals1010, 2020-10-29:
I have try to calibrate imu and camera using the kaliber package.
The issue is the when we run got the error, but first we can try the the demo file given by the dynamic.bag file so its work fine.
I have used this command 
" kalibr_calibrate_imu_camera --target april_6x6_1.yaml --cam camchain-Kalibr_data_2.yaml --imu imu_data_2.yaml --bag Kalibr_data_2.bag --bag-from-to 5 165 "

the the log is 
"
importing libraries
Initializing IMUs:
  Update rate: 100.0
  Accelerometer:
    Noise density: 0.01 
    Noise density (discrete): 0.1 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.005
    Noise density (discrete): 0.05 
    Random walk: 4e-06
Initializing imu rosbag dataset reader:
	Dataset:          Kalibr_data_2.bag
	Topic:            /android/imu
Traceback (most recent call last):
  File "/home/vishal/Desktop/ros_wk/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/vishal/Desktop/ros_wk/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/vishal/Desktop/ros_wk/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 151, in main
    estimateTimedelay=parsed.estimate_imu_delay) )
  File "/home/vishal/Desktop/ros_wk/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 592, in __init__
    parsed.bag_from_to, parsed.perform_synchronization)
  File "/home/vishal/Desktop/ros_wk/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 33, in initImuBagDataset
    perform_synchronization=perform_synchronization)
  File "/home/vishal/Desktop/ros_wk/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_common/ImuDatasetReader.py", line 44, in __init__
    self.indices = self. sortByTime(self.indices)
  File "/home/vishal/Desktop/ros_wk/src/kalibr-master/aslam_offline_calibration/kalibr/python/kalibr_common/ImuDatasetReader.py", line 55, in sortByTime
    topic, data, stamp = self.bag._read_message(self.index[idx].position)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 989, in _read_message
    return self._reader.seek_and_read_message_data_record(position, raw)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 2478, in seek_and_read_message_data_record
    msg_type = _get_message_type(connection_info)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1574, in _get_message_type
    message_type = genpy.dynamic.generate_dynamic(info.datatype, info.msg_def)[info.datatype]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/dynamic.py", line 159, in generate_dynamic
    for line in msg_generator(msg_context, spec, search_path):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/generator.py", line 807, in msg_generator
    genmsg.msg_loader.load_depends(msg_context, spec, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 349, in load_depends
    return load_msg_depends(msg_context, spec, msg_search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 318, in load_msg_depends
    depspec = load_msg_by_type(msg_context, resolved_type, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 125, in load_msg_by_type
    file_path = get_msg_file(package_name, base_type, search_path)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/msg_loader.py", line 84, in get_msg_file
    % (base_type, package, search_path), base_type, package, search_path)
genmsg.msg_loader.MsgNotFound: Cannot locate message [Header]: unknown package [std_msgs] on search path [{}]
"
 the files are like
1 : april_6x6_1.yaml
"
#example for aprilgrid
  target_type: 'aprilgrid' #gridtype
  tagCols: 6                  #number of apriltags
  tagRows: 6                  #number of apriltags
  tagSize: 0.024              #size of apriltag, edge to edge [m]
  tagSpacing: 0.3125             #ratio of space between tags to tagSize

"
2 : camchain-Kalibr_data_2.yaml
"
cam0:
  cam_overlaps: []
  camera_model: pinhole
  distortion_coeffs: [0.03898830865334542, -0.06542439193369343, -0.010361690213818296,
    0.002750578353580892]
  distortion_model: radtan
  intrinsics: [456.6416528387091, 461.90899632373333, 321.0637656581371, 223.35658415094662]
  resolution: [640, 480]
  rostopic: /camera/image

"

3: imu_data_2.yaml
"
rostopic: /android/imu
update_rate: 100.0 #Hz

accelerometer_noise_density: 0.01 #continous
accelerometer_random_walk: 0.0002 
gyroscope_noise_density: 0.005 #continous
gyroscope_random_walk: 4.0e-06

"
4: info Kalibr_data_2.bag 
"
rosbag info Kalibr_data_2.bag 
path:        Kalibr_data_2.bag
version:     2.0
duration:    2:49s (169s)
start:       Oct 27 2020 18:46:03.90 (1603804563.90)
end:         Oct 27 2020 18:48:53.29 (1603804733.29)
size:        2.5 GB
messages:    19732
compression: none [2892/2892 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu   [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /android/imu    16840 msgs    : sensor_msgs/Imu  
             /camera/image    2892 msgs    : sensor_msgs/Image
"
thanks in advance

xudooo, 2020-12-15:
Hi, did you solve this problem?

DandyLorenz, 2022-03-23:
Hi, did you solve this problem?

goldbattle, 2022-04-22:
Looks like you need to source your ROS workspace first.

closed 2022-05-03 by goldbattle