ethz-asl/kalibr #394: The meaning of the report.pdf

TouchDeeper, 2020-11-02:
Hello,
I want to understand the meaning of each graph in the report.pdf. There are some questions:
1.Why are there lines between points?
[image]

2. what is the meaning of the number in this graph?
[image]

NikolausDemmel, 2020-11-03:
> 1\.Why are there lines between points?

I think it's simply a line plot of the residuals sorted by polar angle,

> 2\.  what is the meaning of the number in this graph?

These are just pixel coordinates of your images. I think it visualizes which parts of the image are covered by the observations used during calibration (you want good coverage of all areas).

TouchDeeper, 2020-11-04:
@NikolausDemmel 

> I think it's simply a line plot of the residuals sorted by polar angle,

But I don't understand why there are lines between points, the reporjection error should be for a point.

> These are just pixel coordinates of your images. I think it visualizes which parts of the image are covered by the observations used during calibration (you want good coverage of all areas).

But in the image used for calibration, the calibration target is not as big as the above image

NikolausDemmel, 2020-11-04:
> But I don't understand why there are lines between points, the reporjection error should be for a point.

It's just a line plot of the individual data points. I don't think there is extra meaning (I think).

> But in the image used for calibration, the calibration target is not as big as the above image

Looks to me like these are the patterns detected for ALL images used in calibration. You can somewhat make out the individual grids. If those are much bigger than you would expect from the images, then something went wrong I guess...

TouchDeeper, 2020-11-04:
> It's just a line plot of the individual data points. I don't think there is extra meaning (I think).

Thank you, I want to know what the line between two points meaning, if it has no meaning, there should be just points in the graph.

> Looks to me like these are the patterns detected for ALL images used in calibration. You can somewhat make out the individual grids. If those are much bigger than you would expect from the images, then something went wrong I guess...

But the calibration result is close to the ground truth.  It is strange.

floriantschopp, 2020-11-07:
1. I agree with @NikolausDemmel. I also think the lines have no specific meaning.
2. This is essentially your image (in pixel values) with all corner/grid observations on it (each observation gets a different color). As @NikolausDemmel said, the goal is to have good coverage of observations on the whole image. If your calibration target was way smaller than that, something went wrong in your calibration process. But also, if it is way smaller than that, I think you should get closer or use a bigger target ;-)

TouchDeeper, 2020-11-07:
@NikolausDemmel  @floriantschopp Thanks a lot for your reply.
1. If lines have no specific meaning, why point A link to point B, not point C.
2. I check my image again. I think your reply is right. Thank you.

NikolausDemmel, 2020-11-07:
> If lines have no specific meaning, why point A link to point B, not point C.

I think it's simply all observations sorted by polar angle (angle of viewing ray from the principle axis) and then plotted as a line plot. So A is connected to B because they are the reprojection errors of two pattern points consecutive in that (partial) ordering.

TouchDeeper, 2020-11-07:
> I think it's simply all observations sorted by polar angle (angle of viewing ray from the principle axis) and then plotted as a line plot. So A is connected to B because they are the reprojection errors of two pattern points consecutive in that (partial) ordering.

Thanks a lot for your reply. I think I got it.

closed 2020-11-07 by TouchDeeper