ethz-asl/kalibr #395: So why the results of y and z axis are reverse?

opendvd, 2020-11-09:
I use Kailbr to calibrate the RealSenseD435i, especially the external parameters between IMU and camera. But the external parameters of translation between IMU and camera confused me. The kalibr's results are different with ideal value.
The ideal value is: 
t_IC0 = [-0.006 0.005 0.012]T;
t_IC1 = [0.044 0.005 0.012]T; 
But kalibr result is:
t_IC0 = [-0.00510796 0.01859231 0.00576145]T;
t_IC1 = [0.04383316 0.01886102 0.00574488]T;
So why the results of y and z axis are reverse?

ynma-hanvo, 2021-06-29:
can someone please advice  
first please confirm me if the +x is right, +y down, +z forward, am i right?

mhyoosefian, 2021-09-19:
> can someone please advice
> first please confirm me if the +x is right, +y down, +z forward, am i right?

Hi. Did you find the answer?!

goldbattle, 2022-05-04:
z should be forward, x to the right and y down if looking from behind the camera.
You should be able to check this with the [kalibr_visualize_calibration](https://github.com/ethz-asl/kalibr/blob/master/aslam_offline_calibration/kalibr/python/kalibr_visualize_calibration) script

closed 2022-06-08 by goldbattle