ethz-asl/kalibr #396: Support for Ubuntu 20.04?

jeff-delaune, 2020-11-10:
Hello,

Are there plans to support the latest Ubuntu 20.04.1 release, which comes with OpenCV 4 and Boost 1.71? Compilation currently crashes.

niciBume, 2020-11-20:
I just used the CDE version on 20.04. I that a work around for your problem?

jeff-delaune, 2020-11-24:
Thanks for the suggestion @niciBume. Using the CDE version on 20.04 leads to same issue as #315.

Back to compiling from source, I have solved a few issues related to OpenCV 4 and python updates. I need to see with my work if I can push them here so everyone can benefit. I am still struggling with the following compilation error though:

`jeff@endurance:~/Software/kalibr_ws$ catkin build -DCMAKE_BUILD_TYPE=Release -j4
--------------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/jeff/Software/kalibr_ws
--------------------------------------------------------------------------------
Build Space:        [exists] /home/jeff/Software/kalibr_ws/build
Devel Space:        [exists] /home/jeff/Software/kalibr_ws/devel
Install Space:      [unused] /home/jeff/Software/kalibr_ws/install
Log Space:          [exists] /home/jeff/Software/kalibr_ws/logs
Source Space:       [exists] /home/jeff/Software/kalibr_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        -j4
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------------
[build] Found '37' packages in 0.0 seconds.                                    
[build] Package table is up to date.                                           
Starting  >>> catkin_simple                                                    
Starting  >>> ethz_apriltag2                                                   
Starting  >>> python_module                                                    
Starting  >>> sm_common                                                        
Finished  <<< catkin_simple                                 [ 0.2 seconds ]    
Starting  >>> sm_logging                                                       
Finished  <<< sm_common                                     [ 0.3 seconds ]    
Starting  >>> sm_random                                                        
Finished  <<< python_module                                 [ 0.2 seconds ]    
Starting  >>> suitesparse                                                      
Finished  <<< ethz_apriltag2                                [ 0.4 seconds ]    
Starting  >>> aslam_time                                                       
Finished  <<< sm_logging                                    [ 0.3 seconds ]    
Starting  >>> opencv2_catkin                                                   
Finished  <<< sm_random                                     [ 0.3 seconds ]    
Starting  >>> sm_boost                                                         
Finished  <<< suitesparse                                   [ 0.3 seconds ]    
Starting  >>> sm_matrix_archive                                                
Finished  <<< aslam_time                                    [ 0.3 seconds ]    
Starting  >>> sm_opencv                                                        
Finished  <<< opencv2_catkin                                [ 0.2 seconds ]    
Starting  >>> sm_property_tree                                                 
Finished  <<< sm_boost                                      [ 0.3 seconds ]    
Starting  >>> numpy_eigen                                                      
Finished  <<< sm_matrix_archive                             [ 0.2 seconds ]    
Starting  >>> sm_eigen                                                         
Finished  <<< sm_opencv                                     [ 0.2 seconds ]    
Starting  >>> sm_timing                                                        
Finished  <<< sm_property_tree                              [ 0.3 seconds ]    
Finished  <<< sm_eigen                                      [ 0.3 seconds ]    
Starting  >>> sm_kinematics                                                    
Finished  <<< sm_timing                                     [ 0.2 seconds ]    
Starting  >>> sparse_block_matrix                                              
Finished  <<< numpy_eigen                                   [ 0.7 seconds ]    
Finished  <<< sm_kinematics                                 [ 0.3 seconds ]    
Starting  >>> aslam_cameras                                                    
Starting  >>> sm_python                                                        
[... 349 lines cut ...]
/home/jeff/Software/kalibr_ws/src/Kalibr/aslam_cv/aslam_imgproc/include/aslam/implementation/OmniUndistorter.hpp:116:34: warning: ‘keypoint’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  116 |       int iu = saturate_cast<int>(keypoint[0] * INTER_TAB_SIZE);
      |                ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
[build]   Warnings:  None.                                                                                                                                                              
[build]   Abandoned: 2 packages were abandoned.                                                                                                                                         
[build]   Failed:    1 packages failed.                                                                                                                                                 
[build] Runtime: 27.3 seconds total.`

hz658832, 2020-12-03:
Hey Jeff, @jeff-delaune 
I think I have figured out the issure, i am guessing newer Eigen has changed the member function in protected.
My solution: change line 466 `protected` into `public` in /usr/include/eigen3/Eigen/src/Core/MatrixBase.h or /usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h.
You may have other problems after this issure, feel free to contact me.
My OS: Ubuntu 20.04
Bests from aachen
Zh

saulzar, 2020-12-15:
Even if it will compile in 20.04 (which I have managed also, eventually). All the python code is python2 which I don't believe is supported under OS noetic anymore?

christian-rauch, 2020-12-15:
Yes, the python 2 code likely will not run on Ubuntu 20.04 / python3. But so will be the OpenCV code which uses parts of the C API in the C++ code and will not compile with OpenCV 4.

I started porting to Ubuntu 20.04 (OpenCV 4, Python 3) (https://github.com/christian-rauch/kalibr/tree/fixes) but are now stuck in boost-python with (https://github.com/ethz-asl/kalibr/issues/404):
```
/usr/include/boost/python/detail/destroy.hpp:20:9: error: ‘Eigen::MatrixBase<Derived>::~MatrixBase() [with Derived = Eigen::Matrix<double, -1, 1>]’ is protected within this context
   20 |         p->~T();
      |         ^
```

saulzar, 2020-12-15:
Ah yes - I saw the same and couldn't solve it. Commented that out in order
to compile, but then couldn't run because of the python3 issue and gave
up...

Is there a reason to call the destructor rather than "delete p"?

christian-rauch, 2020-12-16:
I am not sure why boost implemented it this way, a `delete` should have the same effect. Changing the code on-disk is not an option as this is a system package, which should not be interfered with. If the code has to be patched manually, a custom boost or Eigen version should be provided in the workspace.

The [documentation on this](http://eigen.tuxfamily.org/index.php?title=PythonInteropBoost) seems outdated. Maybe it's worth exploring other options, such as pybind11, for this.

xqms, 2021-01-12:
@christian-rauch Thanks for your work on porting to 20.04!

> I am not sure why boost implemented it this way, a delete should have the same effect.

Calling the destructor explicitly just executes the destructor, leaving uninitialized memory behind - ready to be used again (for example by calling the constructor in place). Calling `delete` will deallocate the memory. I guess they implemented some array logic. `std::vector` uses this technique as well.

The main problem is that `Eigen::MatrixBase` is not a good type to allocate / deallocate. You also cannot keep it in `std::vector` for the same reason. I guess some method in kalibr returns this type, when it should return a more concrete type (like `Eigen::Matrix`).

xqms, 2021-01-17:
For anyone else needing a workaround for running kalibr under 20.04, here's a quick&dirty Dockerfile:
https://gist.github.com/xqms/6cc015bae5222b4bff33cd589e4adc05

xqms, 2021-03-02:
I also pushed my dockerfile to docker hub, so you can launch it using
```bash
cd where/my/bagfiles/are
docker run -it -v $(pwd):/mnt xqms/kalibr
```
and then inside the docker container:
```
. /workspace/devel/setup.bash
cd /mnt
kalibr_calibrate_cameras ...
```

mcamurri, 2021-04-11:
Check this out:
https://github.com/ori-drs/kalibr

DreamWaterFound, 2021-04-14:
> Check this out:
> https://github.com/ori-drs/kalibr

Thank vvvvvvvvvvvvvvvvvvvvery much! It works very well!

wanghaichuan941221, 2021-04-15:
> Check this out:
> https://github.com/ori-drs/kalibr

Hello, I managed to make the project, but still get stuck with the optimization as shown below
[image]

Enes1097, 2021-08-24:
Hez, @wanghaichuan941221 I have the same problem, did you find any way to solve it?

xqms, 2021-08-24:
@Enes1097 I had similar problems with the optimization using the ori-drs fork. Have you tried my docker container from above (https://github.com/ethz-asl/kalibr/issues/396#issuecomment-788968713)?

Enes1097, 2021-08-24:
@xqms not yet, but I will try it out. Thanks :)

JzHuai0108, 2021-09-14:
Hi @Enes1097, I tested the [ori-drs fork](https://github.com/ori-drs/kalibr) with the Kalibr [IMU-camera dynamic calibration sample](https://github.com/ethz-asl/kalibr/wiki/downloads), and it worked almost perfectly despite several innocuous warnings from wxPython as shown below.
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/matplotlib/backends/backend_wx.py", line 784, in _onSize
    self.bitmap = wx.Bitmap(self._width, self._height)
wx._core.wxAssertionError: C++ assertion ""width > 0 && height > 0"" failed at /tmp/pip-install-i3i6lzg4/wxPython/ext/wxWidgets/src/gtk/bitmap.cpp(539) in Create(): invalid bitmap size
Traceback (most recent call last):
```
The calibration results are so close to the those obtained under ubuntu 18 + ros1 melodic.
I am not sure how the optimization stuck issue mentioned by others cropped.

FYI, here is the procedure I build and test the ori-drs fork under ubuntu 20 + ros1 noetic.

## Steps to build [ori-drs/kalibr on noetic](https://github.com/ethz-asl/kalibr/issues/396)
This repo also probably work on melodic as mentioned by [wxmerkt](https://github.com/ethz-asl/kalibr/pull/405)

```
sudo apt update
sudo apt-get install python3-setuptools python3-rosinstall ipython3 libeigen3-dev libboost-all-dev doxygen libopencv-dev ros-noetic-vision-opencv ros-noetic-image-transport-plugins ros-noetic-cmake-modules python3-software-properties software-properties-common libpoco-dev python3-matplotlib python3-scipy python3-git python3-pip libtbb-dev libblas-dev liblapack-dev libv4l-dev python3-catkin-tools python3-igraph libsuitesparse-dev 

pip3 install wxPython
```
If you encounter errors like "E: Unable to locate package python3-catkin-tools",
then setup your sources.list and keys as instructed [here](http://wiki.ros.org/Installation/Ubuntu).

```
mkdir ~/kalibr_ws/src
cd ~/kalibr_ws/src
git clone --recursive https://github.com/ori-drs/kalibr

cd ~/kalibr_ws
source /opt/ros/noetic/setup.bash
catkin init
catkin config --extend /opt/ros/noetic
catkin config --merge-devel # Necessary for catkin_tools >= 0.4.
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release

catkin build -DCMAKE_BUILD_TYPE=Release -j4
```

## Test with camera-IMU calibration sample data

```
source ~/kalibr_ws/devel/setup.bash
cd /path/to/kalibr_dynamic_sample
rosrun kalibr kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45
```

marc-wittwer, 2022-01-06:
> I also pushed my dockerfile to docker hub, so you can launch it using
> 
> ```shell
> cd where/my/bagfiles/are
> docker run -it -v $(pwd):/mnt xqms/kalibr
> ```
> 
> and then inside the docker container:
> 
> ```
> . /workspace/devel/setup.bash
> cd /mnt
> kalibr_calibrate_cameras ...
> ```

Thanks a lot this worked.

I had to set the display env variable to let the docker container create a plot.
https://stackoverflow.com/questions/49169055/docker-tkinter-tclerror-couldnt-connect-to-display/49229627#49229627

```
# Install xhost
 sudo apt-get install x11-xserver-utils

# Allow docker container to use your display
xhost +
```

```
sudo docker run -it --rm -v $(pwd):/mnt -e DISPLAY=unix$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw xqms/kalibr
```

goldbattle, 2022-04-22:
Hi please try PR https://github.com/ethz-asl/kalibr/pull/515 to see if that solves your issue.

closed 2022-05-03 by goldbattle