ethz-asl/kalibr #397: angle of ‘T_ci: (imu0 to cam0) is positive to the reality

Yes-Jumby, 2020-11-25:
here is my result:
Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[ 0.99998198  0.00591562 -0.00102339  0.0535171 ]
 [-0.00372559  0.47780644 -0.87845724 -0.12355092]
 [-0.00470764  0.87844523  0.47781987 -0.0774419 ]
 [ 0.          0.          0.          1.        ]]

I have trans it in angle:
imu to camera

angle(°) =

   61.4565    0.2697   -0.2135

here is my reality coordinate:
[image]

As we can see,if coordinate is all in left hand coordinate,the angleX should be -61.
I guess the result of kalibr is in left hand coordinate.

Is there anything I got is wrong?Give me your hand please!

JulioLP, 2020-12-04:
Hi,

T_ci (imu0 to cam0) transforms points from the reference frame imu0 to the reference frame cam0. You can see this using the [0,0,0] vector in the ref. frame imu0:

p_c = T_ci * p_i = T_ci * [0,0,0,1]' = [0.0535171, -0.12355092,  -0.0774419, 1]'

The resulting p_c is the origin of imu0 expressed in the axes of cam0.

In conclusion, the transformation is correct. If you need more insights about how transformations work, I can refer you to some book chapters about that.

Yes-Jumby, 2020-12-07:
> Hi,
>
> T_ci (imu0 to cam0) transforms points from the reference frame imu0 to the reference frame cam0. You can see this using the [0,0,0] vector in the ref. frame imu0:
>
> p_c = T_ci * p_i = T_ci * [...]

yes,I think i understand the translation vector means,What I mean is only the angle is positive if all is in right-hand coordinate(Translation vector is correct)

JulioLP, 2020-12-14:
You can see this transformation as if you are "moving" the axes of cam to the axes of imu. The rotation that you do for this is of 60 degrees counterclockwise in the camera axis X. This is a positive angle, which is what you have. I don't see anything bad in the transformation matrix.

Yes-Jumby, 2020-12-16:
> You can see this transformation as if you are "moving" the axes of cam to the axes of imu. The rotation that you do for this is of 60 degrees counterclockwise in the camera axis X. This is a positive angle, which is what you have. I don't see anything bad in the transformation matrix.

**T_ci: (imu0 to cam0)  As if I am the axes of imu to the axes of cam.**  Is this right?
（if T_ci(cam0 to imu0),the transformation will be all right,i think.but T_ci is T_ci(imu0 to cam0)）

JulioLP, 2020-12-16:
No. T_ci(imu0 to cam0) describes a movement FROM the camera axes TO the imu axes. However, if it multiplies a point described in imu axes, it transforms it to camera axes. It is a bit difficult convention to follow, but it is correct.

Yes-Jumby, 2020-12-16:
> No. T_ci(imu0 to cam0) describes a movement FROM the camera axes TO the imu axes. However, if it multiplies a point described in imu axes, it transforms it to camera axes. It is a bit difficult convention to follow, but it is correct.

**this is really hard for me to understand,would you like to refer me some book chapters about that ?**

schneith, 2020-12-16:
The framework follows a convention very similar to:

https://github.com/ethz-asl/maplab/wiki/Expressing-frame-transformations-in-code

and 

http://paulfurgale.info/news/2014/6/9/representing-robot-pose-the-good-the-bad-and-the-ugly

Yes-Jumby, 2020-12-17:
> **this is really hard for me to understand,would you like to refer me some book chapters about that ?**

> The framework follows a convention very similar to:
> 
> https://github.com/ethz-asl/maplab/wiki/Expressing-frame-transformations-in-code
> 
> and
> 
> http://paulfurgale.info/news/2014/6/9/representing-robot-pose-the-good-the-bad-and-the-ugly

Yes-Jumby, 2020-12-17:
>

Yes, I understand it now . thanks a lot

Yes-Jumby, 2020-12-17:
> **this is really hard for me to understand,would you like to refer me some book chapters about that ?**

I understand it now.  Thank you for your help .Thanks:)

closed 2020-12-17 by Yes-Jumby