ethz-asl/kalibr #399: Kalibr_bagcreater creates imu0 topic only. [CAN'T FIND /cam0/image_raw in rosbag]

GabHoo, 2020-12-09:
Hi there,

I need to perform a camera calibration and later a IMU-camera calibration but I am having problem with the bagcreater.
I am really new to such thing especially with rosbags and ROS in general.

I recorded video and imu data from my Android smartphone (using marslogger app: https://github.com/OSUPCVLab/mobile-ar-sensor-logger) and I converted them into a rosbag with the given script Kalibr_bagcreater. Problem comes up when calling Kalibr_calibrate_cameras, which can't find /cam0/image_raw in the rosbag just created. 
Dataset folder, as it is written in https://github.com/ethz-asl/kalibr/wiki/bag-format, contains one folder named cam0 with all the frames and imu0.csv with imu datas. If I run rosbag info output.bag this comes out in the console:

path:        output.bag
version:     2.0
duration:    2:26s (146s)
start:       Feb 02 1970 19:03:21.92 (2829801.92)
end:         Feb 02 1970 19:05:48.20 (2829948.20)
size:        5.2 MB
messages:    14629
compression: none [7/7 chunks]
types:       sensor_msgs/Imu [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /imu0   14629 msgs    : sensor_msgs/Imu

As I said I don't know a lot about rosbags but I feel in the voice "topics" there should me cam0/image_raw along with /imu0.
There must be something silly I am missing.

Error when launching Kalibr_calibrator_cameras:
RuntimeError: Could not find topic /cam0/image_raw in /home/gabriel/Desktop/kalibr-cde/android_calibration/output.bag.

Thanks to those who will help.

kenmaro3, 2021-02-22:
@GabHoo hello, I encountered same problem. did you figure out what was the cause of this? thx!

yaelbenoren, 2021-08-22:
I had the same problem. It worked for me when I added the titles ("timestamp", "omega_x"....) and converted the file from tab delimited to coma delimited

PPakalns, 2021-08-24:
CDE packaging is incorrect!!!

Changing bash script to `pushd` whole directory instead of dropping last component of path fixed it for me.

```
#wrap relative paths to absolute
args=()
for var in "$@"
do
    #if this is a file (=exists)
    if [ -f $var ]
    then
        #replace paths
        pushd `dirname $var` > /dev/null
        ABS_PATH=`pwd -P`
        popd > /dev/null
        args+=($ABS_PATH"/"$(basename $var))
    elif [ -d $var ]
    then
        #its a directory
	      pushd $var > /dev/null
        ABS_PATH=`pwd -P`
        popd > /dev/null
        args+=($ABS_PATH)
    else
        #its not a file or directory
        args+=("$var")
    fi
done
```

See directory case.

Changed 	      
```
pushd `dirname $var` > /dev/null
```
To
```
 pushd $var > /dev/null
```

Solin32, 2022-03-15:
I had the same problem and the above two solutions didn't work. Add /. to the command after your folder
`kalibr_bagcreater --folder dataset-dir/. --output-bag awsome.bag`
works for me.

goldbattle, 2022-04-22:
It looks like your bag does not have the image topic in it.
Does @ShurlinZhao solution work for you?
If so this might be a command parsing issue or path problem.

goldbattle, 2022-04-22:
It looks like this is a common issue. See https://github.com/ethz-asl/kalibr/issues/146
Will track the fix in that one.

closed 2022-04-22 by goldbattle