ethz-asl/kalibr #400: Coordinate System of Multi-Camera Calibration

ShrutheeshIR, 2020-12-12:
I am trying to calibrate a 2-camera system, using AprilGrids. 

Though the stereo calibration does not converge, this is the matrix that I get:

initialized baseline between cam0 and cam1 to:

```
[[ 0.86731999  0.10467067 -0.48662108 -0.22621089]
 [-0.11083987  0.9937064   0.01618977  0.00591429]
 [ 0.48525308  0.0398953   0.87346312 -0.09904313]
 [ 0.          0.          0.          1.        ]]
```

When I measure my camera distances physically, I find that camera 1 is roughly ```20cm``` to the right of camera 0, and camera 1 is around  ```10 cm``` behind camera 0, and both cameras are nearly at the same height.

Thus, looking at this, it looks like +ve x is to the left, +ve z is forward. What coordinate system is this exactly? What does +ve y correspond to?
It neither seems to follow ROS coordinate system, nor does it follow opencv convention, since OpenCV is +ve x to the right.

How do I get this matrix in OpenCV convention? (i.e. +ve x to the right, +ve y up and +ve z forward).

Thanks,

andrew-t97, 2021-04-15:
Hi ShrutheeshIR, did you ever figure this out?

Thanks

ynma-hanvo, 2021-06-29:
i am also not clear on this.    I was thinking of +x to the right,  +y to the down,   +z  forward,  
but seems like the element at (2, 3)   should be 0.09904313  if cam1 behind cam0