ethz-asl/kalibr #401: OverflowError: bad numeric conversion: positive overflow

TouchDeeper, 2020-12-12:
Hello,
I use `kalibr_calibrate_imu_camera` to calibrate the infra1 and imu of realsense d435i, but an `OverflowError` occurs.

the command is below:
```
kalibr_calibrate_imu_camera --bag imu_infra_left.bag --cam camchain-multi_camera1.yaml --imu imu.yaml --target ../kalibr_target/april_6x6_50x50cm.yaml
```

An `OverflowError` occurs:
```
importing libraries
Initializing IMUs:
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.0261 
    Noise density (discrete): 0.369109739779 
    Random walk: 0.000657
  Gyroscope:
    Noise density: 0.00296
    Noise density (discrete): 0.0418607214462 
    Random walk: 1.7e-05
Initializing imu rosbag dataset reader:
	Dataset:          imu_infra_left.bag
	Topic:            /imu
	Number of messages: 18891
Reading IMU data (/imu)
Traceback (most recent call last):
  File "/home/td/slam/tools/kalibr_ws/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/td/slam/tools/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 236, in <module>
    main()
  File "/home/td/slam/tools/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 151, in main
    estimateTimedelay=parsed.estimate_imu_delay) )
  File "/home/td/slam/tools/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 603, in __init__
    self.loadImuData()
  File "/home/td/slam/tools/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 631, in loadImuData
    for timestamp, omega, alpha in self.dataset:
  File "/home/td/slam/tools/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImuDatasetReader.py", line 20, in next
    return self.dataset.getMessage(idx)
  File "/home/td/slam/tools/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImuDatasetReader.py", line 115, in getMessage
    timestamp = acv.Time( data.header.stamp.secs, data.header.stamp.nsecs )
OverflowError: bad numeric conversion: positive overflow
```

One of the imu topic detail is below, Is the number of  stamp overflow?:
```
header: 
  seq: 71350
  stamp: 
    secs: 3215513653
    nsecs: 159665108
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0086845420301
  y: 0.0680479183793
  z: 0.137869596481
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.979500270589
  y: -9.62279948923
  z: 1.14039308731
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
```
the bag information is:
```
path:        imu_infra_left.bag
version:     2.0
duration:    1:33s (93s)
start:       Dec 12 2020 15:12:12.34 (1607757132.34)
end:         Dec 12 2020 15:13:45.92 (1607757225.92)
size:        590.0 MB
messages:    20879
compression: none [663/663 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu   [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /imu          18891 msgs    : sensor_msgs/Imu  
             /infra_left    1988 msgs    : sensor_msgs/Image
```

The `imu.yaml` is:
```
#Accelerometers
accelerometer_noise_density: 2.61e-02   #Noise density (continuous-time)
accelerometer_random_walk:   6.57e-04   #Bias random walk

#Gyroscopes
gyroscope_noise_density:     2.96e-03   #Noise density (continuous-time)
gyroscope_random_walk:       1.70e-05   #Bias random walk

rostopic:                    /imu      #the IMU ROS topic
update_rate:                 200.0      #Hz (for discretization of the values above)
```

the `camchain-multi_camera1.yaml ` is:
```
cam0:
  cam_overlaps: [1, 2]
  camera_model: pinhole
  distortion_coeffs: [0.003022672106865009, 0.009522466105805063, 0.002871855121949081,
    0.00013495045969115333]
  distortion_model: radtan
  intrinsics: [384.45179643973404, 384.78523859874434, 322.2480691442894, 246.4911371288146]
  resolution: [640, 480]
  rostopic: /infra_left
cam1:
  T_cn_cnm1:
  - [0.9999998198847818, -0.0003722274318073672, -0.00047082601545923473, -0.0501447035432903]
  - [0.00037286937012517076, 0.9999990001295269, 0.0013640778472011302, -0.0002423858184168682]
  - [0.0004703177975003564, -0.0013642531581097783, 0.9999989588066999, -0.0010626486472629568]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [0, 2]
  camera_model: pinhole
  distortion_coeffs: [0.0033094627508105895, 0.0028500564784450853, 0.001308633880489102,
    0.0004804464806785611]
  distortion_model: radtan
  intrinsics: [383.79939184713237, 383.95551659425746, 322.6554246231207, 246.17350031270774]
  resolution: [640, 480]
  rostopic: /infra_right
cam2:
  T_cn_cnm1:
  - [0.9999393449888674, 0.011008057990416787, 0.00035916920802902576, 0.06825152877110491]
  - [-0.011009955374239813, 0.9999225914834031, 0.005795854016294327, 0.0019383166143685088]
  - [-0.0002953403081580662, -0.005799456905656629, 0.9999831393947137, -0.0034999986242408646]
  - [0.0, 0.0, 0.0, 1.0]
  cam_overlaps: [0, 1]
  camera_model: pinhole
  distortion_coeffs: [0.11866656652181382, -0.2409958407252894, 0.0015550868815829427,
    0.0003769207570705737]
  distortion_model: radtan
  intrinsics: [603.2381336187116, 602.97121984669, 330.0316867557939, 245.35437838229035]
  resolution: [640, 480]
  rostopic: /color
```
Could you please provide some advice? Thank you a lot!

hanchengzhe, 2020-12-25:
Hi, have you solved this problem yet? I have the same error.

TouchDeeper, 2020-12-25:
@hanchengzhe This is indeed because of the timestamp of IMU. If you use realsense D435i,  a quick fix is to set the parameter `initial_rest` to true, you can refer to my [doc](https://github.com/TouchDeeper/BetaGo/blob/ztd/betago_calibration/doc/rs_d435i_calibration.md#infra1-infra2-imu-joint-calibration)  about realsense d435i calibration using Kalibr.

martinakos, 2021-02-23:
If I add --perform-synchronization to the kalibr_calibrate_imu_camera launch command the OverflowError disappears for me.

closed 2022-04-22 by goldbattle