ethz-asl/kalibr #404: install error

Jxpenggf, 2020-12-15:
/usr/local/include/eigen3/Eigen/src/Core/util/Macros.h:885:23: error: ‘Eigen::MatrixBase<Derived>::~MatrixBase() [with Derived = Eigen::Matrix<double, -1, 1>]’ is protected
     EIGEN_DEVICE_FUNC ~Derived() = default;
/usr/include/boost/python/detail/destroy.hpp:33:9: error: within this context
         p->~T();
         ^
How to slove it

Xiangyf18, 2021-01-20:
I sovle it finally  by removing  eigen3 and  boost  (which were both installed from source code ) and  then , use “sudo apt-get install libeigen3-dev libboost-all-dev ” to reinstall them .

XFFer, 2022-03-22:
You can find a file named 'MatrixBase.h' in the path '/usr/local/include/eigen3/Eigen/src/Core/' or '/usr/include/eigen3/Eigen/src/Core/'. (I find it in the first place.)
Then, the key step is to modify 'protected' to 'public' on line 466.

```
protected: EIGEN_DEFAULT_COPY_CONSTRUCTOR(MatrixBase) 
                 EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR(MatrixBase)
```
TO
```
public: EIGEN_DEFAULT_COPY_CONSTRUCTOR(MatrixBase)
           EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR(MatrixBase)
```

readsa, 2022-04-22:
This change is submitted in Eigen **3.3.8**.
A temporary workaround is: 
1. Download eigen repository **older** than 3.3.8. 
2. Install to a specified directory, like: /path/to/old/eigen.
3. When building kalibr, add this cmake arg:  -DCMAKE_FIND_ROOT_PATH=/path/to/old/eigen
The build command is like:
catkin build -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_ROOT_PATH=/path/to/old/eigen -j4

goldbattle, 2022-04-22:
Can you see if PR https://github.com/ethz-asl/kalibr/pull/515 fixes this issue for you?

readsa, 2022-04-22:
@goldbattle Cool! It works for me. (Ubuntu 18.04, with opencv4 and eigen 3.4.0 installed)

closed 2022-05-04 by goldbattle