ethz-asl/kalibr #406: question about csv modification

huanhexiao, 2020-12-27:
hi guys, i want to use my own imu0.csv to creat bag via "kalibr_bagcreater --folder ", it shows error:
======
importing libraries
Traceback (most recent call last):
  File "/home/huanhexiao/WORK/Tools/OpenSource_Tools/Kalibr/devel/bin/kalibr_bagcreater", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/huanhexiao/WORK/Tools/OpenSource_Tools/Kalibr/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_bagcreater", line 124, in <module>
    bag.close()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 415, in close
    self._stop_writing()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1264, in _stop_writing
    self.flush()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 285, in flush
    self._stop_writing_chunk()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1330, in _stop_writing_chunk
    self._write_connection_index_record(connection_id, entries)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1403, in _write_connection_index_record
    buffer.write(_pack_time  (entry.time))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1601, in _pack_time
    def _pack_time(v):   return _pack_uint32(v.secs) + _pack_uint32(v.nsecs)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbag/bag.py", line 1599, in _pack_uint32
    def _pack_uint32(v): return struct.pack('<L', v)
struct.error: 'L' format requires 0 <= number <= 4294967295
===
could anyone do me a favour?

goldbattle, 2022-04-27:
Can you please provide the csv which has issues?

vswdigitall, 2022-05-26:
I have same error.
Have recorded 3000 png frames with 60 fps and mpu9250 gyro and accel data.
Bag file created but with same error.
After rosbag reindex data.bag i can see images in rviz. But imu data can't.

vswdigitall, 2022-05-26:
Should be imu0 timestamps same as image file names and imu0.csv lines equal files number? Or ros will sync them in runtime?

vswdigitall, 2022-05-27:
Fixed. It was bad image filename. 
[image3](https://cam360.tv/etc/bag3.jpg)
At the bottom, selected line.

closed 2022-06-02 by goldbattle

Wanghy118, 2022-06-20:
hi, i met the same problem ,that "struct.error : integer out of range for L format code", could you help me ?

vswdigitall, 2022-06-20:
First you can check all timestamps has same length. In imu.csv and png file names.