ethz-asl/kalibr #407: Calibrate camera imu fails - There must be at least one time segment

RotorBoy2020, 2021-01-05:
Hello, 

Iam trying to calibrate my Intel Realsense D435 stereo camera with a pixhawk using the kalibr_calibrate_imu_camera function. Previously, I was able to successfully calibrate my stereo camera alone, but now when I run the kalibr_calibrate_imu_camera function, the following output is returned:

```
importing libraries
Initializing IMUs:
  Update rate: 50.0
  Accelerometer:
    Noise density: 0.05 
    Noise density (discrete): 0.353553390593 
    Random walk: 0.001
  Gyroscope:
    Noise density: 0.02
    Noise density (discrete): 0.141421356237 
    Random walk: 4e-05
Initializing imu rosbag dataset reader:
	Dataset:          /home/rhb/Downloads/kalibr_20062014/kalibr-cde/dynamic/cali-bag3.bag
	Topic:            /mavros/imu/data_raw
	Number of messages: 3056
Reading IMU data (/mavros/imu/data_raw)
  Read 3056 imu readings over 61.1 seconds                    
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.02 [m]
    Spacing 0.006 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [388.97429653087977, 387.9214697175403]
  Principal point: [319.04915523340543, 247.44212402690985]
  Distortion model: equidistant
  Distortion coefficients: [0.3327988694258969, 0.21516467438628134, 0.06749360362702811, -1.3654007062662439]
  baseline: no data available
Camera chain - cam1:
  Camera model: pinhole
  Focal length: [389.5552048459842, 388.4850815392296]
  Principal point: [319.041844046899, 247.6363593768363]
  Distortion model: equidistant
  Distortion coefficients: [0.3270199958143655, 0.17333353903336415, -0.02511201978133221, 0.609208329238125]
  baseline: [[ 0.99999728  0.00004798  0.00232997 -0.0484981 ]
 [-0.00004701  0.99999991 -0.00041736 -0.00009871]
 [-0.00232999  0.00041725  0.9999972   0.00010174]
 [ 0.          0.          0.          1.        ]]
Initializing camera rosbag dataset reader:
	Dataset:          /home/rhb/Downloads/kalibr_20062014/kalibr-cde/dynamic/cali-bag3.bag
	Topic:            /realsense/cam0_image
	Number of images: 1834
Extracting calibration target corners
  Extracted corners for 1830 images (of 1834 images)                              
Initializing camera rosbag dataset reader:
	Dataset:          /home/rhb/Downloads/kalibr_20062014/kalibr-cde/dynamic/cali-bag3.bag
	Topic:            /realsense/cam1_image
	Number of images: 1833
Extracting calibration target corners
  Extracted corners for 1829 images (of 1833 images)                              
Baseline between cam0 and cam1 set to:
T=  [[ 0.99999728  0.00004798  0.00232997 -0.0484981 ]
 [-0.00004701  0.99999991 -0.00041736 -0.00009871]
 [-0.00232999  0.00041725  0.9999972   0.00010174]
 [ 0.          0.          0.          1.        ]]
Baseline:  0.0484983108431  [m]

Building the problem
	Spline order: 6
	Pose knots per second: 70
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (m/s^2): -1.000000
	Gyroscope Huber width (rad/s): -1.000000
	Do time calibration: False
	Max iterations: 30
	Time offset padding: 0.020000

Estimating imu-camera rotation prior

Initializing a pose spline with 0 knots (100.000000 knots per second over 0.000000 seconds)
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 206, in <module>
  File "<string>", line 172, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 113, in buildProblem
    self.CameraChain.findOrientationPriorCameraChainToImu(self.ImuList[0])
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 438, in findOrientationPriorCameraChainToImu
    self.camList[0].findOrientationPriorCameraToImu( imu )
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 122, in findOrientationPriorCameraToImu
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 276, in initPoseSplineFromCamera
    pose.initPoseSplineSparse(times, curve, knots, 1e-4)
RuntimeError: [Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/bsplines/src/BSpline.cpp:970: initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]: There must be at least one time segment
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/apport_python_hook.py", line 64, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
ImportError: No module named apport.fileutils

Original exception was:
Traceback (most recent call last):
  File "/var/kalibr-build/devel/bin/kalibr_calibrate_imu_camera", line 5, in <module>
    exec(fh.read())
  File "<string>", line 206, in <module>
  File "<string>", line 172, in main
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccCalibrator.py", line 113, in buildProblem
    self.CameraChain.findOrientationPriorCameraChainToImu(self.ImuList[0])
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 438, in findOrientationPriorCameraChainToImu
    self.camList[0].findOrientationPriorCameraToImu( imu )
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 122, in findOrientationPriorCameraToImu
    poseSpline = self.initPoseSplineFromCamera( timeOffsetPadding=0.0 )
  File "/var/kalibr-build/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 276, in initPoseSplineFromCamera
    pose.initPoseSplineSparse(times, curve, knots, 1e-4)
RuntimeError: [Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/bsplines/src/BSpline.cpp:970: initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]: There must be at least one time segment 
```

The key line seems to be: 

```
RuntimeError:` [Exception] /var/kalibr-build/src/kalibr/aslam_nonparametric_estimation/bsplines/src/BSpline.cpp:970: initSplineSparse() assert(numSegments >= 1) failed [0 >= 1]: There must be at least one time segment
```

I am quite confused, as both the stereo camera images and the IMU are time stamped. Any suggestions on how to remedy this issue? I have seen other posts refer to this issue, but have not seen a solution posted yet.

Any help would be greatly appreciated! Thank you

RotorBoy2020, 2021-01-19:
Any help would be greatly appreciated

floriantschopp, 2021-01-21:
Hey @RotorBoy2020 
This quite looks like your images (time stamped on the realsense) and your IMU measurements (time stamped on the Pixhawk) do not share the same clock and that the two clocks have a huge time offset resulting in non-overlapping time stamps. You can check that by [restamping your bag](https://github.com/ethz-asl/dataset_tools/blob/master/python/restamp_bag.py) and then look at `rqt_bag` to see if the two modalities are actually overlapping (as they should be).

feng899, 2021-01-24:
Now, would you like to ask if this problem has been solved?

goldbattle, 2022-04-22:
PX4 doesn't provide valid timestamps. Camera and IMU need to be in the same clock frame (e.g. the ROS clock). Please inspect the raw timestamps. You can see the issue in this message:
```
Initializing a pose spline with 0 knots (100.000000 knots per second over 0.000000 seconds)
```

closed 2022-06-16 by goldbattle