ethz-asl/kalibr #408: Extrinsic transformation matrix (T_cn_cnm1)

mdasari823, 2021-01-10:
Hi, I am trying to calibrate a Stereo camera and confused about the transformation matrix (T_cn_cnm1). In this description, it is given that "_camera extrinsic transformation, always with respect to the last camera in the chain (e.g. cam1: T_cn_cnm1 = T_c1_c0, takes cam0 to cam1 coordinates)_". I need a transformation matrix from the other way i.e., say T_cn_cnm0, where the transformation should be with respect to the first camera (e.g. cam0: T_cn_cnm0 = T_c0_c1, takes cam1 to cam0 coordinates).

I just multiplied the translation vector (i.e., 4th column) with -1. It is working somewhat close to what I expected. However, I am not sure if this is the correct way. Do I have to do something for the rotation matrix as well? Thanks.

ynma-hanvo, 2021-06-29:
mulitply by -1 with the translation  is the same as   inverse()  result , so u can inverse first, then use the translation part;
stereo camera are supposed to have no rotation between left and right ( am I right ?); 
so i guess no need for rotation matrix ;

goldbattle, 2022-05-19:
This has some details if you are not familiar with frame transformations:
https://ingmec.ual.es/~jlblanco/papers/jlblanco2010geometry3D_techrep.pdf

closed 2022-05-19 by goldbattle