ethz-asl/kalibr #412: Question: gravity vector in target coords

chunibyo-wly, 2021-01-24:
Hello, thanks for your work?
What should I suppose to get (`gravity vector in target coords`) when I put an april tag in this orientation
[image]

`axis Y` is up
`axis X` is right
`axis Z` is to person

I think the `gravity vector.y` is supposed to be around `-9.8`, am I right?

thanks for your reply.

_Originally posted by @chunibyo-wly in https://github.com/ethz-asl/kalibr/issues/86#issuecomment-766313491_

lapchun, 2021-04-09:
I'm also confused about this question.

mhyoosefian, 2021-09-19:
Hi. Yeah, the gravity vector will be `[0; -9.81; 0]` ideally if the april tag is placed vertically.

chunibyo-wly, 2021-09-24:
thanks for you reply, and i'll close this issue

closed 2021-09-24 by chunibyo-wly