ethz-asl/kalibr #413: Kalibr gives wrong transformation

mertturanli, 2021-02-02:
Hello,

I am trying to perform camera-imu calibration with kalibr but the optimization fails after 30 iterations (max iterations).

Here is the output of the node:

```
Initializing IMUs:
  Update rate: 200
  Accelerometer:
    Noise density: 0.00222129 
    Noise density (discrete): 0.0314137844396 
    Random walk: 7.82e-06
  Gyroscope:
    Noise density: 0.00019724
    Noise density (discrete): 0.00278939483042 
    Random walk: 2.29e-06
Initializing imu rosbag dataset reader:
	Dataset:          /home/mert/vio-files/imu-rosbag/2021-01-26-11-41-28_kalibr.bag
	Topic:            /imu/data
bagstart 1611650489.46
baglength 237.564305067
[ WARN] [1612262135.700157]: BagImuDatasetReader: truncated 5664 / 41375 messages.
	Number of messages: 41375
Reading IMU data (/imu/data)
  Read 35711 imu readings over 205.0 seconds                         
Initializing calibration target:
  Type: aprilgrid
  Tags: 
    Rows: 9
    Cols: 7
    Size: 0.05 [m]
    Spacing 0.01 [m]
Initializing camera chain:
Camera chain - cam0:
  Camera model: pinhole
  Focal length: [1057.6455345108845, 706.4478734119666]
  Principal point: [942.6541461598658, 415.8409580811273]
  Distortion model: radtan
  Distortion coefficients: [-0.2926907946425598, 0.07338520501936205, -0.002725873009452831, 0.006160374301270044]
  baseline: no data available
Initializing camera rosbag dataset reader:
	Dataset:          /home/mert/vio-files/imu-rosbag/2021-01-26-11-41-28_kalibr.bag
	Topic:            /camera/image_raw
[ WARN] [1612262143.358643]: BagImageDatasetReader: truncated 656 / 4755 images.
	Number of images: 4755
Extracting calibration target corners
  Extracted corners for 4099 images (of 4099 images)                              

Building the problem
	Spline order: 6
	Pose knots per second: 100
	Do pose motion regularization: False
		xddot translation variance: 1000000.000000
		xddot rotation variance: 100000.000000
	Bias knots per second: 50
	Do bias motion regularization: True
	Blake-Zisserman on reprojection errors -1
	Acceleration Huber width (sigma): -1.000000
	Gyroscope Huber width (sigma): -1.000000
	Do time calibration: True
	Max iterations: 30
	Time offset padding: 0.030000
Estimating time shift camera to imu:

Initializing a pose spline with 20490 knots (100.000000 knots per second over 204.899486 seconds)
  Time shift camera to imu (t_imu = t_cam + shift):
-0.1951448355759936

Estimating imu-camera rotation prior

Initializing a pose spline with 20490 knots (100.000000 knots per second over 204.899486 seconds)
Gravity was intialized to [-9.8050185   0.13141683  0.11298079] [m/s^2]
  Orientation prior camera-imu found as: (T_i_c)
[[-0.9997874  -0.00158617 -0.02055795]
 [ 0.02057167 -0.00918419 -0.9997462 ]
 [ 0.00139696 -0.99995657  0.00921487]]
  Gyro bias prior found as: (b_gyro)
[0.00148579 0.00096441 0.00051469]

Initializing a pose spline with 20502 knots (100.000000 knots per second over 205.019486 seconds)

Initializing the bias splines with 10251 knots

Adding camera error terms (/camera/image_raw)
  Added 4099 camera error terms                               

Adding accelerometer error terms (/imu/data)
  Added 35661 of 35711 accelerometer error terms (skipped 50 out-of-bounds measurements)

Adding gyroscope error terms (/imu/data)
  Added 35661 of 35711 gyroscope error terms (skipped 50 out-of-bounds measurements)
```

The output of the optimization is below:

```

[0.0]: J: 1.88393e+07
Last step was a regression. Reverting
[1]: J: 5.37582e+07, dJ: -3.49189e+07, deltaX: 0.715823, LM - lambda:10 mu:2
[2]: J: 1.04315e+06, dJ: 1.77962e+07, deltaX: 0.189999, LM - lambda:40 mu:4
Last step was a regression. Reverting
[3]: J: 3.83223e+06, dJ: -2.78908e+06, deltaX: 0.457818, LM - lambda:13.3333 mu:2
[4]: J: 944943, dJ: 98208, deltaX: 0.0652052, LM - lambda:53.3333 mu:4
Last step was a regression. Reverting
[5]: J: 2.2378e+06, dJ: -1.29286e+06, deltaX: 0.307365, LM - lambda:17.7778 mu:2
[6]: J: 941532, dJ: 3411.23, deltaX: 0.0299404, LM - lambda:71.1111 mu:4
Last step was a regression. Reverting
[7]: J: 1.12086e+06, dJ: -179324, deltaX: 0.197452, LM - lambda:23.7037 mu:2
[8]: J: 939538, dJ: 1994, deltaX: 0.0159738, LM - lambda:94.8148 mu:4
Last step was a regression. Reverting
[9]: J: 960491, dJ: -20953.4, deltaX: 0.120796, LM - lambda:31.6049 mu:2
[10]: J: 938656, dJ: 882.225, deltaX: 0.00880803, LM - lambda:126.42 mu:4
[11]: J: 936814, dJ: 1842.13, deltaX: 0.0715676, LM - lambda:42.1399 mu:2
...
```

The data is collected with low exposure and with good illumination.

The resulting translation is wrong, it is near zero.

As far as I can see from the discussions on the internet, the delay of the imu-camera may be a problem which is about 200 ms in my case.

What can be the problem?

Thanks in advance.

Best regards.

mertturanli, 2021-02-05:
I managed to find a way to reduce the camera time delay. The results get a lot better. 
I think that the problem is solved, so the issue can be closed.
```

Transformation T_cam0_imu0 (imu0 to cam0, T_ci): 
[[-0.99962891  0.00861574 -0.02584216  0.05306187]
 [ 0.02570054 -0.01616167 -0.99953903 -0.01523521]
 [-0.00902942 -0.99983227  0.01593425 -0.16779918]
 [ 0.          0.          0.          1.        ]]

cam0 to imu0 time: [s] (t_imu = t_cam + shift)
0.003556920556781823
```

Best regards.

closed 2021-02-05 by mertturanli

theDimitron, 2021-08-30:
hello mertturanli,

could you please tell, how exactly did you reduce the camera time delay?

mt-ant, 2021-08-30:
Hi theDimitron,

Mine was a exposure issue.
I reduced the exposure of the camera and gave extra light to the april grid. That solved the camera delay problem.

Best Regards,
Mert Turanli

mertturanli, 2021-08-30:
Hi theDimitron,

Mine was an exposure issue.
I reduced the exposure of the camera and gave extra light to the april grid. That solved the camera delay problem.

Best Regards,
Mert Turanli