ethz-asl/kalibr #418: fix bug information

snowflowersnowflake, 2021-02-25:
=> Processing catkin package: 'sm_boost'
==> Creating build directory: 'build_isolated/sm_boost'
==> Building with env: '/home/y00/kalibr_workspace/install_isolated/env.sh'
==> cmake /home/y00/kalibr_workspace/src/kalibr/Schweizer-Messer/sm_boost -DCATKIN_DEVEL_PREFIX=/home/y00/kalibr_workspace/devel_isolated/sm_boost -DCMAKE_INSTALL_PREFIX=/home/y00/kalibr_workspace/install_isolated -G Ninja in '/home/y00/kalibr_workspace/build_isolated/sm_boost'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/y00/kalibr_workspace/devel_isolated/sm_boost
-- Using CMAKE_PREFIX_PATH: /home/y00/kalibr_workspace/install_isolated;/opt/ros/kinetic
-- This workspace overlays: /home/y00/kalibr_workspace/install_isolated;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/y00/kalibr_workspace/build_isolated/sm_boost/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
CMake Error at /home/y00/kalibr_workspace/install_isolated/share/sm_common/cmake/sm_commonConfig.cmake:110 (message):
  Project 'sm_common' specifies 'include' as an include dir, which is not
  found.  It does not exist in
  '/home/y00/kalibr_workspace/install_isolated/include'.  Report the problem
  to the maintainer 'Paul Furgale ' and request to
  fix the problem.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:4 (find_package)

Marine98k, 2022-03-27:
do you  have solve  this problem

goldbattle, 2022-04-22:
Does PR https://github.com/ethz-asl/kalibr/pull/515 fix this issue for you?

closed 2022-05-03 by goldbattle