ethz-asl/kalibr #425: Wrong  translation result

goktugyildirim, 2021-03-17:
The tool gives a transformation matrix result with the wrong translation. Translations are always very close to zero. What is the reason for this?

Transformation T_cam0_imu0 (imu0 to cam0, T_ci): [m]
[[ 0.1779013   0.98379231 -0.02244609  0.00026128]
 [ 0.27785686 -0.07210108 -0.95791284  0.00006366]
 [-0.94400566  0.16417714 -0.28618032  0.00015286]
 [ 0.          0.          0.          1.        ]]

wengyaokun, 2021-10-06:
Have you found a solution? I have the same problem。

RuanJiaJun, 2021-12-16:
I also encountered this problem. Have you solved it now? Can you share the solution? Thank you！

wengyaokun, 2021-12-17:
The time stamp interval must be uniform. For example, the IMU time stamp interval of 100 Hz should be about 10 ms

goktugyildirim, 2021-12-17:
This tool may work well, but nobody knows how to do :)

KBhanup, 2022-01-31:
> The time stamp interval must be uniform. For example, the IMU time stamp interval of 100 Hz should be about 10 ms

I think by now you might have figured out your issue @wengyaokun; I am assuming you are talking about Px4 imu publish rate and the imu rate in (100hz) imu.yaml file right??!

I did check those and tried to set them same; but still for some reason my transitional values are wrong they are like  [x,y,z; 0.0601, -0.0039, -1.13 m] while actually they would be something like [12cm, 0.3cm and  6cm] also my Gyro mean(3.7) and Accelerometer std(265.0) values are way too high..

I am unable to trace the reason why its going wrong !! Could you please point me out in that direction? Pleaseee!!

goldbattle, 2022-04-22:
Please provide an example bag and more details on what sensors and how the data is collected.

goktugyildirim, 2022-05-04:
I think its expected result since translation error accumulation. IMU cannot give position change directly without error even making camera trajectory continuous using konitiki. So, rotation between camera and IMU can be provided accurately using gyro, but no translation. Also, I wouldn't like to rely upon IMU drivers, low pass filter can be applied.

goktugyildirim, 2022-05-04:
> Please provide an example bag and more details on what sensors and how the data is collected.

I tried to calibrate SBG and usb camera without any hardware based time sync. I have no data currently.

goldbattle, 2022-05-04:
With a usb camera you most likely have a rolling shutter camera (as
compared to a global shutter one).
This causes additional issues.
Kalibr uses a continuous-time formulation.
Hope this helps.

closed 2022-06-16 by goldbattle