ethz-asl/kalibr #430: 1 package failed and 2 packages were abandoned:Failed     << aslam_cv_python:make

hu-minghao, 2021-03-31:
Hello, I follow the instruction with ubuntu16.04 for installation, and aslam_cv_python fail to build . this is the message

`Finished  <<< incremental_calibration_python                [ 0.4 seconds ]                                                                                         
Starting  >>> aslam_splines_python                                                                                                                                  
Finished  <<< aslam_cv_error_terms                          [ 0.3 seconds ]                                                                                         
Finished  <<< aslam_splines_python                          [ 0.4 seconds ]                                                                                         
____________________________________________________________________________________________________________________________________________________________________
Errors     << aslam_cv_python:make /home/nvidia/kalibr_ws/logs/aslam_cv_python/build.make.002.log                                                                   
In file included from /usr/local/include/eigen3/Eigen/Core:88:0,
                 from /home/nvidia/kalibr_ws/src/kalibr/Schweizer-Messer/numpy_eigen/include/numpy_eigen/boost_python_headers.hpp:23,
                 from /home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cv_python/src/CameraProjections.cpp:1:
/usr/include/boost/python/detail/destroy.hpp: In instantiation of ‘static void boost::python::detail::value_destroyer<false>::execute(const volatile T*) [with T = Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >]’:
/usr/include/boost/python/detail/destroy.hpp:95:15:   required from ‘void boost::python::detail::destroy_referent_impl(void*, T& (*)()) [with T = const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >]’
/usr/include/boost/python/detail/destroy.hpp:101:26:   required from ‘void boost::python::detail::destroy_referent(void*, T (*)()) [with T = const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >&]’
/usr/include/boost/python/converter/rvalue_from_python_data.hpp:135:51:   required from ‘boost::python::converter::rvalue_from_python_data<T>::~rvalue_from_python_data() [with T = const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >&]’
/usr/include/boost/python/converter/arg_from_python.hpp:107:8:   required from ‘PyObject* boost::python::detail::caller_arity<2u>::impl<F, Policies, Sig>::operator()(PyObject*, PyObject*) [with F = bool (aslam::cameras::DoubleSphereProjection<aslam::cameras::NoDistortion>::*)(const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >&) const; Policies = boost::python::default_call_policies; Sig = boost::mpl::vector3<bool, aslam::cameras::DoubleSphereProjection<aslam::cameras::NoDistortion>&, const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >&>; PyObject = _object]’
/usr/include/boost/python/object/py_function.hpp:38:33:   required from ‘PyObject* boost::python::objects::caller_py_function_impl<Caller>::operator()(PyObject*, PyObject*) [with Caller = boost::python::detail::caller<bool (aslam::cameras::DoubleSphereProjection<aslam::cameras::NoDistortion>::*)(const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >&) const, boost::python::default_call_policies, boost::mpl::vector3<bool, aslam::cameras::DoubleSphereProjection<aslam::cameras::NoDistortion>&, const Eigen::MatrixBase<Eigen::Matrix<double, -1, 1> >&> >; PyObject = _object]’
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cv_python/src/CameraProjections.cpp:475:1:   required from here
/usr/local/include/eigen3/Eigen/src/Core/util/Macros.h:885:23: error: ‘Eigen::MatrixBase<Derived>::~MatrixBase() [with Derived = Eigen::Matrix<double, -1, 1>]’ is protected
     EIGEN_DEVICE_FUNC ~Derived() = default;
                       ^
/usr/local/include/eigen3/Eigen/src/Core/MatrixBase.h:468:5: note: in expansion of macro ‘EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR’
     EIGEN_DEFAULT_EMPTY_CONSTRUCTOR_AND_DESTRUCTOR(MatrixBase)
     ^
In file included from /usr/include/boost/python/converter/rvalue_from_python_data.hpp:10:0,
                 from /usr/include/boost/python/converter/registry.hpp:9,
                 from /usr/include/boost/python/converter/registered.hpp:8,
                 from /usr/include/boost/python/object/make_instance.hpp:10,
                 from /usr/include/boost/python/object/make_ptr_instance.hpp:8,
                 from /usr/include/boost/python/to_python_indirect.hpp:11,
                 from /usr/include/boost/python/converter/arg_to_python.hpp:10,
                 from /usr/include/boost/python/call.hpp:15,
                 from /usr/include/boost/python/object_core.hpp:14,
                 from /usr/include/boost/python/args.hpp:25,
                 from /usr/include/boost/python.hpp:11,
                 from /home/nvidia/kalibr_ws/src/kalibr/Schweizer-Messer/numpy_eigen/include/numpy_eigen/boost_python_headers.hpp:24,
                 from /home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cv_python/src/CameraProjections.cpp:1:
/usr/include/boost/python/detail/destroy.hpp:33:9: error: within this context
         p->~T();
         ^
make[2]: *** [CMakeFiles/aslam_cv_python.dir/src/CameraProjections.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/DoubleSphereProjection.hpp:289:0,
                 from /home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras.hpp:11,
                 from /home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cv_python/src/CameraGeometries.cpp:2:
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/DoubleSphereProjection.hpp: In instantiation of ‘void aslam::cameras::DoubleSphereProjection<DISTORTION_T>::euclideanToKeypointDistortionJacobian(const Eigen::MatrixBase<OtherDerived>&, const Eigen::MatrixBase<OtherDerived>&) const [with DERIVED_P = Eigen::Matrix<double, 3, 1>; DERIVED_JD = Eigen::Matrix<double, -1, -1>; DISTORTION_T = aslam::cameras::NoDistortion]’:
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/CameraGeometry.hpp:688:9:   required from ‘void aslam::cameras::CameraGeometry<PROJECTION_T, SHUTTER_T, MASK_T>::euclideanToKeypointIntrinsicsJacobian(const Vector3d&, Eigen::MatrixXd&, bool, bool, bool) const [with PROJECTION_T = aslam::cameras::DoubleSphereProjection<aslam::cameras::NoDistortion>; SHUTTER_T = aslam::cameras::GlobalShutter; MASK_T = aslam::cameras::NoMask; Eigen::Vector3d = Eigen::Matrix<double, 3, 1>; Eigen::MatrixXd = Eigen::Matrix<double, -1, -1>]’
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cv_python/src/CameraGeometries.cpp:182:1:   required from here
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/DoubleSphereProjection.hpp:508:42: warning: unused parameter ‘p’ [-Wunused-parameter]
     const Eigen::MatrixBase<DERIVED_P> & p,
                                          ^
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/DoubleSphereProjection.hpp: In instantiation of ‘void aslam::cameras::DoubleSphereProjection<DISTORTION_T>::euclideanToKeypointDistortionJacobian(const Eigen::MatrixBase<OtherDerived>&, const Eigen::MatrixBase<OtherDerived>&) const [with DERIVED_P = Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::Block<const Eigen::Matrix<double, 4, 1>, 3, 1, false> >; DERIVED_JD = Eigen::Matrix<double, -1, -1>; DISTORTION_T = aslam::cameras::NoDistortion]’:
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/DoubleSphereProjection.hpp:550:42:   required from ‘void aslam::cameras::DoubleSphereProjection<DISTORTION_T>::homogeneousToKeypointDistortionJacobian(const Eigen::MatrixBase<OtherDerived>&, const Eigen::MatrixBase<OtherDerived>&) const [with DERIVED_P = Eigen::Matrix<double, 4, 1>; DERIVED_JD = Eigen::Matrix<double, -1, -1>; DISTORTION_T = aslam::cameras::NoDistortion]’
/home/nvidia/kalibr_ws/src/kalibr/aslam_cv/aslam_cameras/include/aslam/cameras/implementation/CameraGeometry.hpp:723:9:   required from ‘void aslam::cameras::CameraGeometry<PROJECTION_T, SHUTTER_T, MASK_T>::homogeneousToKeypointIntrinsicsJacobian(const Vector4d&, Eigen::MatrixXd&, bool, bool, bool) const [with PROJECTION_T = aslam::cameras::DoubleSphereProjection<aslam::cameras::NoDistortion>; SHUTTER_T = aslam::cameras:: [... cut ...]
[build]   Warnings:  None.                                                                                                                                          
[build]   Abandoned: 2 packages were abandoned.                                                                                                                     
[build]   Failed:    1 packages failed.                                                                                                                             
[build] Runtime: 1 minute and 34.6 seconds total.  `

eigen version is 3.2.29 and python is 2.7.

hu-minghao, 2021-03-31:
`No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.6 LTS
Release:	16.04
Codename:	xenial
nvidia@tegra-ubuntu:~/kalibr_ws$ cat /proc/cpuinfo | grep name | cut -f2 -d: | uniq -c
      1  ARMv8 Processor rev 3 (v8l)
      2  ARMv8 Processor rev 0 (v8l)
      3  ARMv8 Processor rev 3 (v8l)
`
this is information of system

fightmoney, 2021-04-11:
I have the same question with you ,have you finished it?

KopiSoftware, 2021-04-13:
Commen that line works, but it may raise some problems about memories. Have a nice day.

goldbattle, 2022-04-22:
Is this still an issue when you use PR https://github.com/ethz-asl/kalibr/pull/515 ?

hu-minghao, 2022-04-22:
你好，已收到，谢谢。

closed 2022-06-02 by goldbattle

hu-minghao, 2022-06-02:
你好，已收到，谢谢。