ethz-asl/kalibr #434: camera reprojection error

whxtsangle, 2021-05-07:
this is my camera yaml：
cam0:
  cam_overlaps: [1, 3]
  camera_model: pinhole
  intrinsics: [9.3660058585368881e+02, 9.3890336908692325e+02, 6.6479012977060790e+02, 3.6434391054746698e+02]
  distortion_model: radtan
  distortion_coeffs: [-3.5031386384876573e-01, 1.0918098703559266e-01,1.1990132999065205e-03,
    -2.3021036530497512e-03]
  
  
  resolution: [1280, 720]
  rostopic: /cam0/image_raw

The camera was calibrated by Zhang Zhengyou calibration method

Calibrated camera and imu
I got a result as follow:

Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0):     mean 163.21369517, median 147.090306184, std: 97.6578719436
Gyroscope error (imu0):        mean 3.78627429466, median 2.48722354141, std: 5.96304399826
Accelerometer error (imu0):    mean 1401.19884122, median 1472.95004308, std: 614.453727896

Residuals
----------------------------
Reprojection error (cam0) [px]:     mean 163.21369517, median 147.090306184, std: 97.6578719436
Gyroscope error (imu0) [rad/s]:     mean 2.57020822002e-06, median 1.68838332716e-06, std: 4.04784849378e-06
Accelerometer error (imu0) [m/s^2]: mean 5.9447832145, median 6.24919778285, std: 2.60690638632

Transformation (cam0):
-----------------------
T_ci:  (imu0 to cam0): 
[[ 0.19315709  0.98013119  0.04509096  0.68180311]
 [ 0.92802758 -0.19742164  0.31589478 -0.17730054]
 [ 0.31852026 -0.01917166 -0.94772216  4.75999742]
 [ 0.          0.          0.          1.        ]]

T_ic:  (cam0 to imu0): 
[[ 0.19315709  0.92802758  0.31852026 -1.48331091]
 [ 0.98013119 -0.19742164 -0.01917166 -0.6120024 ]
 [ 0.04509096  0.31589478 -0.94772216  4.5364202 ]
 [ 0.          0.          0.          1.        ]]

timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-0.334255993062

Gravity vector in target coords: [m/s^2]
[-4.97943415  7.83171422 -3.16826622]

Calibration configuration
=========================

cam0
-----
  Camera model: pinhole
  Focal length: [936.6005858536888, 938.9033690869232]
  Principal point: [664.7901297706079, 364.343910547467]
  Distortion model: radtan
  Distortion coefficients: [-0.3503138638487657, 0.10918098703559266, 0.0011990132999065205, -0.002302103653049751]
  Type: aprilgrid
  Tags: 
    Rows: 6
    Cols: 6
    Size: 0.02 [m]
    Spacing 0.006 [m]

IMU configuration
=================

IMU0:
----------------------------
  Model: calibrated
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.0003 
    Noise density (discrete): 0.00424264068712 
    Random walk: 3e-06
  Gyroscope:
    Noise density: 4.8e-08
    Noise density (discrete): 6.78822509939e-07 
    Random walk: 5e-10
  T_i_b
    [[ 1.  0.  0.  0.]
     [ 0.  1.  0.  0.]
     [ 0.  0.  1.  0.]
     [ 0.  0.  0.  1.]]
  time offset with respect to IMU0: 0.0 [s]

The Reprojection error is wrong.
Could you please help to analyze what may be the problem there.

pkrush, 2021-08-24:
This is really high: 
timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-0.334255993062

goldbattle, 2022-04-22:
I recommend recollecting another dataset and ensuring you have good calibration board extraction.

closed 2022-04-22 by goldbattle