ethz-asl/kalibr #435: Ubuntu 18.04 python-igraph --upgrade error

seungsunkim, 2021-05-14:
I am trying to install Kalibr on Ubuntu 18.04.

I ran sudo pip install python-igraph --upgrade
I get the following error.

Is there a solution?

[image]

twetto, 2021-05-15:
python-igraph drops support below python 3.6 since version 0.9.0, so you may want to download the [older 0.8.3 code](https://github.com/igraph/python-igraph/releases/tag/0.8.3) and run `python setup.py install` yourself.

seungsunkim, 2021-05-17:
@twetto 
The python-igraph version is already installed as 0.7.1.post6-5

aahuhu, 2021-05-28:
this problem has been solved when i ran ''pip install python-igragh''.

closed 2022-04-22 by goldbattle