ethz-asl/kalibr #436: how to use imu-calibr.yaml to calibrate imu instrinsic

arenas7307979, 2021-05-17:
hii, 
Thank your for sharing this great tool, 
I have a question about scale-misalignment. 

I ready know IMU intrinsic have Misalignment, scale, Bias  from imu_tk
[ https://github.com/Neil-Oyoung/imu_tk]

How to use kalibr internal parameters to calibrate the IMU instinsic like imu_tk？ 

[image]

mhyoosefian, 2021-12-03:
Hi,

See [this](https://github.com/ethz-asl/kalibr/issues/189#issuecomment-978890978) comment.

Regards,
Mohammad

goldbattle, 2022-04-22:
This paper has details of the different models:
https://timohinzmann.com/publications/icra_2016_rehder.pdf

For example you can see the following:
[image]