ethz-asl/kalibr #437: extrinsic matrix & kalibr coordinate system

wildecat73, 2021-05-19:
I'm working an a 3D rendering project where 2 cameras are used. I need to transform 3D positions obtained with respect to "cam0" into the coordinate system of "cam1", and then do some rendering of these 3D positions with the OpenGL API. We decided to use Kalibr to obtain the extrinsic transformation between cam0 and cam1. Unfortunately, I have no prior experience with Kalibr, and while I managed to get an extrinsic calibration, I have problems applying it. This might be due to a calibration error, or it might be due to me not applying the extrinsic matrix correctly. I'm using Kalibr via docker image (windows PC), so I do not get any fancy graphics at the end of the multi-camera kalibration, just some text files.

this is the matrix i get via the camchain.yaml:
  T_cn_cnm1:
  - [0.9977650988674991, -0.001967386701723313, 0.06679024533196266, -0.0010258575112577994]
  - [0.0207102620818789, 0.9594478212560744, -0.2811244623498279, 0.1293575924310637]
  - [-0.06352867483615386, 0.2818794204558853, 0.9573442953279789, -0.039399480440791]
  - [0.0, 0.0, 0.0, 1.0]
 
Question 1: 
Is the 4x4 matrix needed to transform a vertex/position from cam0 to cam1 the matrix above, or is it the inverse of the matrix?

Question 2:
The project makes heavy use of 3D rendering with OpenGL - I must know how the Kalibr coordinate system relates to the default OpenGL coordinate system. Initially, I assumed that Kalibr uses the same coordinate system as OpenCV, since the OpenCV lib is used by Kalibr. But now I'm not so sure any more. An image showing the Kalibr coordinate axes would be greatly appreciated.

codegrafix, 2021-08-24:
Did you find answers to your questions? I am working on a similar task. Thanks!

goldbattle, 2022-08-18:
This transform is the Cn to Cn+1 transform. Meaning if you had a 3d point in the Cn frame, you multiply it by this SE(3) transform to get the point in the Cn+1 frame. It has the rotation Cn to Cn+1 and position of Cn in Cn+1. Kalibr uses right handed coordinate systems, so besides this change you should be able to use this transform directly. Hope this helps.

closed 2022-08-18 by goldbattle