ethz-asl/kalibr #440: what is the purpose of imu noise density bias random walk for ?

Ashing00, 2021-06-17:
Hi,

may I know what is the purpose of imu noise density bias random walk for ?
and this configuration ( imu.yaml) be used once or everytime I need to re-create it for different camera? 

 for Camera-IMU calibration ,do we need to do IMU calibration ready firstly,right?

gotiobg, 2021-06-18:
Noise density and ARW (Angular Random Walk) are used to calibrate the IMU.
The values can be used every time if you had already calibrated them for a long enough time, say two hours.

You can check https://github.com/gaowenliang/imu_utils as it gives you these needed params in order to use kalibr.
Here is a nice guide for MPU6050 https://automaticaddison.com/visualize-imu-data-using-the-mpu6050-ros-and-jetson-nano/

Ashing00, 2021-06-24:
hi gotiobg 
got it ,thank you

gotiobg, 2021-06-24:
You're welcome.

closed 2022-04-22 by goldbattle