ethz-asl/kalibr #441: ImportError

wuyang171, 2021-06-17:
$ kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45
importing libraries
Traceback (most recent call last):
  File "/home/yang/opencv2/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yang/opencv2/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 3, in <module>
    import sm
  File "/home/yang/opencv2/devel/lib/python2.7/dist-packages/sm/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/yang/opencv2/devel/lib/python2.7/dist-packages/numpy_eigen/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
ImportError: /home/yang/opencv2/devel/lib/python2.7/dist-packages/numpy_eigen/libnumpy_eigen.so: undefined symbol: _ZTIN5boost6python17error_already_setE

gotiobg, 2021-06-18:
Have you installed boost on your system?

wuyang171, 2021-06-18:
> Have you installed boost on your system?

yes,i did.and when i resolve the relationship,i don't know what's wrong with it.
```
$ ldd libnumpy_eigen.so  
	linux-vdso.so.1 (0x00007ffc6f1d6000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f812b5b6000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f812b22d000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f812b015000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f812ac24000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f812aa07000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f812a7e8000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f812a5e4000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f812a3e1000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f812a043000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f812c18e000)
```
any idea?

gotiobg, 2021-06-18:
Try these
- https://github.com/calamares/calamares/issues/1242#issuecomment-527867365
- https://github.com/BVLC/caffe/issues/2348#issuecomment-241953745

wuyang171, 2021-06-20:
> Try these
> 
>     * [calamares/calamares#1242 (comment)](https://github.com/calamares/calamares/issues/1242#issuecomment-527867365)
> 
>     * [BVLC/caffe#2348 (comment)](https://github.com/BVLC/caffe/issues/2348#issuecomment-241953745)

i tyied,but here comes new problem.and i am new to python,any idea?thanks a lot.

```
$ kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45
importing libraries
Warning: the package aslam_cv_python is not compiled. Type 'rosmake aslam_cv_python' if you need this.
Traceback (most recent call last):
  File "/home/yang/slam/kalibr_ws/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yang/slam/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 4, in <module>
    import kalibr_common as kc
  File "/home/yang/slam/kalibr_ws/devel/lib/python2.7/dist-packages/kalibr_common/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 3, in <module>
  File "/home/yang/slam/kalibr_ws/src/Kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py", line 8, in <module>
    import aslam_cv_backend as cvb
  File "/home/yang/slam/kalibr_ws/devel/lib/python2.7/dist-packages/aslam_cv_backend/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 17, in <module>
  File "<string>", line 18, in Omni
AttributeError: 'module' object has no attribute 'OmniCameraGeometry'
```

wuyang171, 2021-06-20:
menwhile i can't install python-igraph ,i did sudo pip3 install python-igraph --upgrade,but here comes problem with sudo pip install python-igraph --upgrade
```

Collecting python-igraph
  Downloading python-igraph-0.9.6.tar.gz (3.6 MB)
     |████████████████████████████████| 3.6 MB 362 kB/s 
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-NDAd_1/python-igraph/setup.py'"'"'; __file__='"'"'/tmp/pip-install-NDAd_1/python-igraph/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-gTp1IP
         cwd: /tmp/pip-install-NDAd_1/python-igraph/
    Complete output (6 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-NDAd_1/python-igraph/setup.py", line 465
        raise RuntimeError(f"Failed to clean {folder} with git")
                                                              ^
    SyntaxError: invalid syntax
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

gotiobg, 2021-06-20:
Both errors seem to come from outdated python version. Try updating python 2 or use python 3 instead.

wuyang171, 2021-06-21:
> Both errors seem to come from outdated python version. Try updating python 2 or use python 3 instead.

i used python3 to catkin_make it,it seems didn't work well.i think i need to learn how to debug python.i couldn't understand the error.

```
kalibr_calibrate_imu_camera --target april_6x6.yaml --cam camchain.yaml --imu imu_adis16448.yaml --bag dynamic.bag --bag-from-to 5 45
importing libraries
Initializing IMUs:
  Update rate: 200.0
  Accelerometer:
    Noise density: 0.01 
    Noise density (discrete): 0.141421356237 
    Random walk: 0.0002
  Gyroscope:
    Noise density: 0.005
    Noise density (discrete): 0.0707106781187 
    Random walk: 4e-06
Initializing imu rosbag dataset reader:
	Dataset:          dynamic.bag
	Topic:            /imu0
Traceback (most recent call last):
  File "/home/yang/kalibr/kalibr_ws/devel/bin/kalibr_calibrate_imu_camera", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/yang/kalibr/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 245, in <module>
    main()
  File "/home/yang/kalibr/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_imu_camera", line 153, in main
    estimateTimedelay=parsed.estimate_imu_delay) )
  File "/home/yang/kalibr/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 592, in __init__
    parsed.bag_from_to, parsed.perform_synchronization)
  File "/home/yang/kalibr/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_imu_camera_calibration/IccSensors.py", line 33, in initImuBagDataset
    perform_synchronization=perform_synchronization)
  File "/home/yang/kalibr/kalibr_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_common/ImuDatasetReader.py", line 27, in __init__
    self.bag = rosbag.Bag(bagfile)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 467, in __init__
    self._open(f, mode, allow_unindexed)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 1427, in _open
    if   mode == 'r': self._open_read(f, allow_unindexed)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 1463, in _open_read
    self._reader.start_reading()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosbag/bag.py", line 2674, in start_reading
    raise ROSBagUnindexedException()
rosbag.bag.ROSBagUnindexedException: Unindexed bag
```

gotiobg, 2021-06-21:
http://wiki.ros.org/rosbag/Troubleshooting

wuyang171, 2021-06-23:
> http://wiki.ros.org/rosbag/Troubleshooting

thanks a lot.it finally worked in python3.7 and catkin make -DBoost_NO_BOOST_CMAKE=ON,have a nice day!

gotiobg, 2021-06-23:
You're welcome. You too.

DandyLorenz, 2022-03-22:
> i used python3 to catkin_make it,it seems didn't work well.i think i need to learn how to debug python.i couldn't understand the error.
>
> ```
> kalibr_calibrate_imu_camera --target april_6x6.yaml --cam [...]
I got the same error when I perform joint calibration of camera and imu, can you share how you solved it? thank you very much.

DandyLorenz, 2022-03-26:
yes,Currently I have solved it.

DandyLorenz, 2022-03-26:
> > http://wiki.ros.org/rosbag/Troubleshooting
> 
> thanks a lot.it finally worked in python3.7 and catkin make -DBoost_NO_BOOST_CMAKE=ON,have a nice day!

It's a really bad decision and it's wreaking havoc on my ROS.ros melodic usually uses python2.7, and higher versions of python3 are not supported. If you upgrade, normal ros functions may not be available, such as catkin_make. I tried to upgrade python3 and my ROS system was reinstalled. The author's error has been resolved so far. It is about the reason why Android recorded the bag file itself.

closed 2022-04-22 by goldbattle